opentx/opentx #8192: Opentx Companion crash on simulation with Big Sur 11.1

peroed, 2021-01-03:
Opentx 2.3.10
MacOS 11.1/Imac 27 late 2015
Transmitter Taranis Q X7 Access.

Whenever I start simulation, companion immediately crashes.  Regardless of model i chose.

RAV3NFPV, 2021-01-07:
I am actually kind of in the same boat, I just got the new Radiomaster TX16s and got opentx set up and working fine. When I get on the companion on my pc and try to SIMULATE a model, I get this error message: "SIMULATOR FOR THIS FIRMWARE NOT YET AVAILABLE." Never had this issue on my QX7.

3djc, 2021-01-07:
> I am actually kind of in the same boat, I just got the new Radiomaster TX16s and got opentx set up and working fine. When I get on the companion on my pc and try to SIMULATE a model, I get this error message: "SIMULATOR FOR THIS FIRMWARE NOT YET AVAILABLE." Never had this issue on my QX7. Not sure how to fix this nor find any help online.

Your issue as nothing in common with peroed one, so why post on his issue ? As for your issue, please go to community sites like rcgroups or openrcforum for end user support

RAV3NFPV, 2021-01-07:
> Your issue as nothing in common with peroed one, so why post on his issue ? As for your issue, please go to community sites like rcgroups or openrcforum for end user support

Why post? Anyone with eyes can see why I did. Simulate issues with opentx but I'd be happy to go check another forum, preferably one that has members that have a little bit more respect than you.
This part of your comment , "Your issue as nothing in common with peroed one, so why post on his issue ?" came off a bit dickish. Could've been a little bit more friendly about it instead of making me look like I'm incompetent on finding the correct area to post my radio issue......

kilrah, 2021-01-07:
This issue is about "crash on simulation with Big Sur 11.1", not "any problem about simulation"...

You also didn't give any information related to your setup, what companion version whether you're on mac os big sur like posting this here would suggest,

peroed, 2021-01-08:
I was able to narrow the problem down.

Companion works fine if I select any transmitter but QX7 Access. QX7 works, X9D+works. Only QX7 Access makes companion crash when I start simulation.
Crash Report attached
[Crash Report Companion.txt](https://github.com/opentx/opentx/files/5785440/Crash.Report.Companion.txt)

3djc, 2021-01-08:
We do not have Macs so a bit hard for us to work on this.  But looking at the crash log, we made changes to this area already. Could  you try a nightly companion ? (https://downloads.open-tx.org/2.3/nightlies/companion/macosx/opentx-companion-2.3.11N409.dmg)

And also specify if it happens with a blanki model or with a specific model

peroed, 2021-01-08:
Installed the nightly. No change.

Happens if I select empty document without models and if I create a generic model.

Let me know if I can do anything from my side.

3djc, 2021-01-08:
We may have found something, will need a new nightly so you can test, stay tuned

3djc, 2021-01-08:
@peroed could you try : https://downloads.open-tx.org/2.3/nightlies/companion/macosx/opentx-companion-2.3.11N410.dmg ?

peroed, 2021-01-08:
Works!! Thanks for the effort.

3djc, 2021-01-08:
Thanks for the tests ! 👍

closed 2021-01-08 by 3djc