opentx/opentx #8200: Cannot flash with dfu-util 0.10

nmaggioni, 2021-01-08:
**Describe the bug**

Writing a new firmware on the radio fails if `dfu-util` version 0.10 is used. Version 0.9 is working as expected.

Manually executing `/usr/bin/dfu-util -a 0 --dfuse-address 0x08000000:524288 --device 0483:df11  -U/tmp/33827-flash-check.bin` in a terminal, copy-pasted from the Configurator's window, works fine even with both `dfu-util` versions.

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure that `dfu-util` 0.10 is installed.
2. Power off and connect the radio.
3. Click on "Write firmware to Radio" in the Configurator's sidebar.
4. Proceed with "Write to TX" without changing any option.
5. Encounter error.
6. Unplug the radio.
7. Downgrade to `dfu-util` 0.9.
8. Repeat steps 2-4.
9. No error should appear.

**Screenshots**

[image]

[image]

**General information**
 - **OpenTX version:** upgrading from 2.3.10 to 2.3.11
 - **Hardware:** FrSky X-Lite S, Horus X10S
 - **OpenTX Companion:** 2.3.11 on Manjaro Linux 20.2.1

3djc, 2021-01-08:
Disable firmware check

eff917, 2021-01-14:
The actual problem is an extra empty parameter before the lat parameter, which in shell translates to a single extra space:
`[D] readFirmware():152 - readFirmware: reading "/home/eff/data/personal/FPV/Radio/TX16S/2.3.11/backup.bin" with "/usr/bin/dfu-util" ("-a", "0", "--dfuse-address", "0x08000000:2097152", "--device", "0483:df11", "", "-U/home/eff/data/personal/FPV/Radio/TX16S/2.3.11/backup.bin")`

PR #8556 (2021-06-17): dfu-util 0.10 compatibility
Fixes #8200 by removing an empty element that generated extra whitespace in the actual command invocation, considered illegal by dfu-util v0.10.

I tried to trace back the addition of the incriminated empty string but the blames stop [here](https://github.com/nmaggioni/opentx/blame/e1b7848a91df7b9c854d68d44c9f8754c40bab6e/companion/src/radiointerface.cpp#L74), giving no evident reason as to why that change was needed. Perhaps an older implicit type conversion issue?

Tested by flashing an Horus X10S with a fresh build of the companion.
closed 2021-06-17 by bsongis