opentx/opentx #8204: Range Check can be select when in RX mode of the Multimodule

wucke13, 2021-01-10:
**Describe the bug**
Using the internal Multimodule of a Radiomaster TX16S, in the Model setup, I can select the menu item for Range Check for the `FrSkyRX` mode. The menu item is not visible, but selectable. Pressing enter on it starts the `RSSI:` Screen.  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to  `Model Setup`
2. Set `Mode` to `MULTI  FrSkyRX  RX`
3. Select `Bind`
4. Scroll down one click

**Expected behavior**
The selection moves from Bind to `RF Freq. fine tune` (without invisible `Range Check` menu item in between).

**Screenshots**
Between all of the screenshots I just scrolled down one element. Observe how no visibile menu item is selected in the second screenshot.

[image]
[image]
[image]

**General information**
 - OpenTX version 2.1.10-otx (7ac1ecb6)
 - Hardware Radiomaster TX16S Hall

3djc, 2021-01-10:
There is no range check for RX protocol...

3djc, 2021-01-10:
ok, I can reproduce

closed 2021-01-15 by bsongis

wucke13, 2021-01-15:
Perfect, thanks!

commit 0931bef6 (2021-01-15): Merge pull request #8208 from opentx/3djc/multi-fixes

Fix #8204
commit 76996342 (2021-01-15): Fix #8204