opentx/opentx #8205: Binding beep keeps going in MULTI RX mode

wucke13, 2021-01-10:
**Describe the bug**
When I select the `MULTI  FrSkyRX  RX` protocol on internal multimodule, the bind sound does not stop after bind mode is deactivated again

**To Reproduce**
Steps to reproduce the behavior:
1. Open `MODEL SETUP`
2. Select `MULTI  FrSkyRX  RX`
3. Activate `Bind` mode, the bind mode is announced through repeating "beep-beep-beep" sounds
4. Deactivate `Bind` mode by pressing enter again on the menu item
5. The "beep-beep-beep" sound keeps going

**Expected behavior**
The "beep-beep-beep" stops after bind mode is deactivated

**General information**
 - OpenTX version 2.1.10-otx (7ac1ecb)
 - Hardware Radiomaster TX16S

3djc, 2021-01-11:
We are working with @pascallanger on this. Stay tuned

3djc, 2021-01-11:
Fixed in upcoming multi v1.3.1.98

closed 2021-01-11 by 3djc