opentx/opentx #8206: clever sd sync/update process in companion

wucke13, 2021-01-10:
**Is your feature request related to a problem? Please describe.**
When updating the OpenTX firmware, one normally as well copies over the content of the new sd-card zip file for the new version. I'm well aware that I could as well manually change the opentx version file, but I do much prefer using the sd card files which are provided by opentx. _Except, for when I changed one file_. For this use case I would be interested in only having a way of updating only those files which where untouched by me.

**Describe the solution you'd like**
1. Every sd-card contents zip file comes with an index (maybe even in the opentx version file, semantically that makes a bit of sense in my opinion) which stores the hash value of each file in the zip.
2. A sd card update procedure is added to companion. The following algorithm is used to determine, whether a file will be updated. The new files are all the files from the updated sd card contents zip. By file path I mean the full path (including file name) but relative to the root of the sd card/zip file. 

for each new file from the zip
  if a file exists on the sd under the new files path
    if said file has the same hash as in the index
       said file will be update with the contents of the new file
    else 
      print file will not be updated because it is modified
  else
    print file will not be updated because it was deleted

Depending on the user feedback, one could as well integrate a dialog based system where the user gets to decide what to do on new files (file exists in zip, but not on sd) and changed files (file on sd was changed).

The end goal would be to allow comfortable update of the sd card using the companion, without manually recreating all contents on the sd card everytime.

**Describe alternatives you've considered**
An alternative would be to use a layered fs approach: A `sd-propagated` folder, and all files in there will be copied automatically to the sd after unpacking the new zip. However, this becomes more tricky for the day to day sd synchronisation, because we somehow would have to decide to which sd folder a file gets. Not impossible, but might bear some nasty edge cases. Also a strategy must be decided on how to represent a file which is to be deleted in said `sd-propagated` folder. Maybe an empty file could help, if there is no use case for dead empty files?

**Additional context**
Managing the sd card is a rather manual process atm. Companion offers basic tools for synchronization, but no real solution for updating the sd card. Making this more convenient is my aim.

lshems, 2021-01-10:
Why not simply a subfolder per version, as the firmware itself is version aware.
KISS.

wucke13, 2021-01-10:
@lshems Can you elaborate a bit what you mean? I'm all in for kiss, but so far have only a very vague idea of what you could mean by that.

elecpower, 2021-01-11:
Companion SD card management could do with some work as it catches out quite a few. I did start working on an idea sometime ago but it came to a halt with other priorities.
One idea is to optionally rename the current SD card structure as a backup, build a new SD card from automated download matching the current radio profile instead of the current manual step, unzip the SD card , optionally only keep the language set for the firmware to reduce the number of files, and merge a new customisations folder structure for the radio profile.
As a separate step use the existing SD card sync process to transfer the updated contents to the radio.

wucke13, 2021-01-11:
I really like the idea of an overlay, for this purpose. If done correctly, that makes it very simple to manage the data.

lshems, 2021-01-11:
Separate opentx updated folders from user custom files and folders.
So media (images, soundfiles, ...) Should be stored in a version linked location, and the user ones on a non-version linked location.

lshems, 2021-01-11:
Don't forget that users want to manage different versions of opentx ON THE RADIO. By using the firmware install from the sdcard, no pc to be used.
In that way we can manage rollbacks within a mayor version easily, and with a complete backup on the radio even over mayor versions.

lshems, 2021-01-11:
I would already be happy with a complete directory structure with head directory named after radiotype, with a sub for major version and another sub for minor version, and then ALL current folders.

The radio would have to use that as location instead of root.

wucke13, 2021-01-11:
> Separate opentx updated folders from user custom files and folders.

I'm in!

> So media (images, soundfiles, ...) Should be stored in a version linked location, and the user ones on a non-version linked location.

Here I disagree. Because both sets are not disjunct, users have custom media files, or delete some, like the `tada` (?) files. Some sort of merge/overlay must be used to allow custom version of the default files.

> Don't forget that users want to manage different versions of opentx ON THE RADIO. 

This is an excellent point of which I have not thought before!

> I would already be happy with a complete directory structure with head directory named after radiotype, with a sub for major version and another sub for minor version, and then ALL current folders.

Here I do not understand what the benefit would be. Please elaborate more about the benefits for this :slightly_smiling_face:

lshems, 2021-01-11:
> > Separate opentx updated folders from user custom files and folders.
> 
> I'm in!
> 
> > So media (images, soundfiles, ...) Should be stored in a version linked location, and the user ones on a non-version linked location.
> 
> Here I disagree. Because both sets are not disjunct, users have custom media files, or delete some, like the `tada` (?) file. Some sort of merge/overlay must be used to allow custom version of the default files.

Manage by priority. No user file means system file. Manage separately. Empty file only name in user part should not trigger error, but skip system file.
> This is an excellent point of which I have not thought before!
>
>
> Here I do not understand what the benefit would be. Please elaborate more about the benefits for this 🙂
If each firmware would look in a radiotype/version/subversion folder for all files, it would allow the above things already to a great extend, even if managed manually.

wucke13, 2021-01-11:
> Manage by priority. No user file means system file. Manage separately. Empty file only name in user part should not trigger error, but skip system file.

Well thats basically what I mean by an overlay fs. Then yes, this seems like a good approach IMO.

> If each firmware would look in a radiotype/version/subversion folder for all files, it would allow the above things already to a great extend, even if managed manually.

Agreed. One thing which bothers me though: Fat32 doesn't have support for symlinks. Duplicating all the files on the SD seems like a bad idea as well. I would propose a thrd layer for this:

+ One folder with base files (I suppose audio files for example do not need to be different on many radios)
+ One folder with device/version specific files (like png vs bmp images, depending on the radios screen)
+ One folder with user specific files

The lookup works from bottom to top in that list: If it doesn't exist in the User folder, than it is looked up in the device specific folder. Only if it is still not found the base folder is considered. From an algorithmic standpoint this seems not too complicated at first glances, depending on the granularity that might change though. However, I'm not convinced that one would want this structure on the SD card itself. After all, this roughly doubles (or tripples for the third layer approach) the sd accesses. I'm eager to find out what the core devs think about adding this extra logic in OpenTX.

lshems, 2021-01-11:
Then you could create a 'join files' function.

Actually, I could manage this all in Lua. Lol.

File copy is possible, though dirty, open and read and open and write all content in a loop. Works fine, but dirty.

lshems, 2021-01-11:
Actually it would perhaps even be preferable in Lua. Just add some basic file handling to the Lua apiy

lshems, 2021-01-11:
Also, consider log file locations. Forgot. No media, but over different versions no need to manage.

lshems, 2021-01-11:
Radio backups and model backups are also only major version relevant. They don't change between minors.

mpjf01, 2021-07-12:
I think there are three categories of file. First, there are those provided by OpenTx by default that will be used by OpenTx unless otherwise instructed. Second, there are those that the user never wants used, eg some users never want to hear the startup sound. Third, there are those supplied by the user that are never to be overwritten by the OpenTx defaults. Could there be some way for the user to identify the second and third category that would be a signal for OpenTx to act differently? Something like a special character, or string, in the file name?

lshems, 2021-07-12:
I think it is obvious from the above that a structural solution is needed. Not a hack based on filenames that is bound to create unwanted errors for users.

mpjf01, 2021-07-12:
And who was it that argued earlier in the thread for keeping it simple?

But no worries, argue for your perceived perfect solution that will never happen because it will take too much scarce development effort.

lshems, 2021-07-12:
Look, I don't need it. 

People seem to forget that there are also suggestions done from a user perspective without having the need yourself.

But I know, most requests from others are just apersonal need.

Please try to be objective and fair. 

This discussion was started because it is currently not user friendly, about which there is little to argue.

That it doesn't need a fix is on the Devs, and it has long been known that they don't give shit about user-friendly behaviour of opentx.

They just want to have fun and enjoy what they are doing .
So don't yell at me please in these kind of wordings. They I'll be responded likewise.

wucke13, 2021-07-13:
For deleting of default files, is there any real objection towards empty files? Like having the default files, and than an overlay with custom user files where empty files just result in a deletion of default files?