opentx/opentx #8207: Global persistent timer

dmmanson, 2021-01-10:
In release 2.3.11 the "statistics" page (previously available in 2.3.10) seems to be missing (or I can no longer find it). That page contained a value for a "global persistent timer." This information could be used to keep track of the total Tx "ON" time since the last battery charge. This feature was mentioned in Issue 4950.

3djc, 2021-01-10:
Radio type ?

dmmanson, 2021-01-10:
FrSky X10S

3djc, 2021-01-10:
This hasn't changed. Long enter then statistics

closed 2021-01-10 by 3djc

dmmanson, 2021-01-10:
Hmm.  Press "SYS" > PgUp/Dn to Info .  No cursor, no "Statistics"
Page name is
 "VERSION" with lines for FW, VERS, DATE, TIME, EEPR, OPTS

[image]

3djc, 2021-01-10:
Where did I talk about SYS ???

dmmanson, 2021-01-10:
You didn't talk about SYS.  My bad.  Let's leave it at that.