opentx/opentx #8209: Checking of checklist against physical items

Supermagnum, 2021-01-11:
Hi I think that it should be possible to get the checklist to monitor the physical items on the radio, like switches, potmeters and so on.

A example is a three position switch.
This toggles spoilers on a fast glider, position 0% , 50% 100%>
If the switch is in the 0% position, a green ✓ appears on the checklist preflight item called Spoilers retracted.
Other items could be model battery voltage,TX battery voltage, items in telemetry as GPS lock, IMU alignment complete and so on.

lshems, 2021-01-11:
Typical Lua task. Would ask for Lua script at startup instead, and you can create everything you want.

Problem might be the switching of the RF module, as that is part of precheck finalisation. Could be done with a Lua function as well though.

Supermagnum, 2021-01-13:
> Typical Lua task. Would ask for Lua script at startup instead, and you can create everything you want.
> 
> Problem might be the switching of the RF module, as that is part of precheck finalisation. Could be done with a Lua function as well though.

I sadly can't program anything.

lshems, 2021-01-13:
Ok, so you talk about a green check, so you have a Horus x12 or similar.

Post a copy of your checklist in text format, and I'll make you a dynamic checklist widget.

It won't prevent your model from being started as in the text checklist that is shown at model load. But that's not what your after I think.

Guido.

Supermagnum, 2021-01-13:
> Ok, so you talk about a green check, so you have a Horus x12 or similar.
>
> Post a copy of your checklist in text format, and I'll make you a dynamic checklist widget.
>
> It won't prevent your model [...]

Here is my basic suggestion :

Preflight checklist:
[] Battery voltage within 15.8 volts
[]Flaps down
[]Aileron movement checklist ( -100 +100 %)
[]rudder check ( -100 +100 %)
[]Elevator check ( -100 +100 %)
[] trottle 0-100 % check
[] RX battery check ( in this case 7.4 V 2600 mAH battery )

Telemetry from https://github.com/yaapu/FrskyTelemetryScript ( Frsky passtrough )
[] telemetry link active
[]GPS lock
[]servo movement check ( assumes the ardupilot is in FBW A mode, pitch and tilt the plane it to confirm that the control surfaces move the correct way to return it to level flight. The ailerons and elevators will move)
[] INS checks (i.e. Acclerometer and Gyro checks) and so on, common pre-arm aka pre flight checks are listed here : https://ardupilot.org/plane/docs/common-prearm-safety-checks.html

It would be nice if the items on the list is not marked with a checkmark if the physical items are not moved.
Perhaps this needs input from the ardupilot/FrskyTelemetryScript developers ?

lshems, 2021-01-13:
Ok, it becomes more clear now. You want to shift the ardupilot checklist to the radio and prevent the necessity to link to the GCS?

Supermagnum, 2021-01-13:
> Ok, it becomes more clear now. You want to shift the ardupilot checklist to the radio and prevent the necessity to link to the GCS?

Yes, I think that it is handy for those who use Ardupilot and radios that can comprehend telemetry from Ardupilot.

lshems, 2021-01-13:
But I think that in that case there are better ways to do things. You don't want to maintain a checklist on two places.

It would be preferable to get the info from ardupilot to the radio, but based on ardupilot doing the checks.

lshems, 2021-01-13:
Check the ArduPilot telemetry sensors that are read on the radio. perhaps you can use one of those to signal the checklist completion or errorcode from the ardupilot. To be discussed with the ardupilot devs.

I'm happy to do things in LUA on the radio, but this is an entirely different task than the title suggest, in my opinion.

It is not related to ON RADIO items, but to all kind of things.

closed 2021-10-12 by stale[bot]