opentx/opentx #8210: Display turns on/off every few seconds after flashing 2.3.11

NameOfTheDragon, 2021-01-12:
**Describe the bug**
After updating formware 2.3.10 -> 2.3.11, LCD screen goes completely dark (no display, no backlight) for several seconds at a time. Then seconds later it will turn back on, pressing any key sends it dark again. It seems impossible to enter a long-press.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash firmware 2.3.11 and power-cycle.
2. After announcing "Welcome to OpenTX" and displaying the logo, radio begins to make voice announcements but after about half a second, the voice output is truncated and the display goes dark.

**Expected behavior**
Obviously, the display turning off every few seconds shouldn't happen

**General information**
 - OpenTX version 2.3.11
 - Hardware Radiomaster TX16S

3djc, 2021-01-12:
Please reflash the firmware from SD card

NameOfTheDragon, 2021-01-13:
> Please reflash the firmware from SD card

I would be very happy to do that, but I'll need some instructions.

- When I select the firmware file in the radio menus, it offers me the option to "flash bootloader" bit not firmware.
- There is no user manual on Radiomaster's web site
- The OpenTX user manual is FrSky specific and I can't find any reference to flashing from SD card
- It shouldn't make any difference where the firmware image comes from, should it?

bigtwin, 2021-02-11:
I have noticed several forums and YT video, which are showing all about Dadiomasters. However, the only forum which I would search for help is a native german but I am sure there are some in UK. They can assist you in flashing Radiomasters.
As far as I know, you need to place a certain file on a certain position and power on the transmitter.

closed 2022-04-17 by stale[bot]