opentx/opentx #8217: (Minor) In Preflight Checks EX1 display label is shortened to EX (EX2 on the other hand is shown correctly as "EX2")

rotorman, 2021-01-14:
**Bug description**

When optional inputs EX1 and EX2 are enabled with setting "Pot with detent" and they are included in model Preflight checks, their state is checked on power-up.
If EX1 is not at the center, a warning is given, but the display text says "EX" without a number. Correct would be "EX1".

(On the other hand EX2 is handled correctly -> the display shows "EX2")

**To Reproduce**
Tested with OpenTX v2.3.11 on RadioMaster TX16S with two additional pots connected to inputs SI & SJ.
1) Enable EX1 & EX2 as "Pot with detent"
2) In model setup, enable under Preflight checks EX1 & EX2
3) power off
4) make sure EX1 and EX2 are not centered
5) power up
6) observe the wrong warning label for EX1

**Expected behavior**
Preflight check dialog should say "EX1" if EX1 is not centered and not just "EX"

**General information**
 - OpenTX version v2.3.11
 - RadioMaster TX16S with additional two pots connected to SI & SJ

3djc, 2021-01-14:
[image]

looks like your description is not quite accurate, it displays ex1 alright, but then overlaps

rotorman, 2021-01-14:
Indeed you are right! I did not even notice the overlaid 1 with my not so young eyes anymore. So, just a positioning/spacing problem.

PR #8218 (2021-01-15): Properly space swich/pots check text
This fixes #8217 

[image]
closed 2021-01-15 by bsongis