opentx/opentx #8220: Models in sdcard\MODELS

sfjuocekr, 2021-01-15:
If you have been mocking about with a bunch of models and have deleted them eventually, there are still modelX.bin files listed for the model on the sdcard.

For example, I have six models but my MODELS folder shows nine. I can tell from the data that it does use the first six models and the other three are just leftovers that never got removed. So, when I make a new model now... it starts at 10 but should start at 7.

kilrah, 2021-01-16:
Intended behavior.

closed 2021-01-16 by kilrah

sfjuocekr, 2021-01-16:
I went through the code and figured it was indeed intended behavior to save writes to SD. The radio actually refers to the first created model, but if you import them in companion it saves them one by one.