opentx/opentx #8223: Add a menu option to set the UART baud rate. Especially, when LUA is selected for serial port.

rotorman, 2021-01-16:
**Describe the solution you'd like**
(With RadioMaster TX16S) under menu SYStem -> Hardware -> Serial port (TTL) or Serial port 2 (TTL) can be assigned to LUA.

If LUA is selected, it would be nice if there would be an additional option to select/set the baud rate for the communication.

[image]

**Describe alternatives you've considered**
Alternatively, there could be a LUA command to set the baud rate.

rotorman, 2021-01-19:
It looks like @3djc just [implemented](https://github.com/opentx/opentx/commit/c38e513c7e3b8cf33d2c48531f6a491ea9aa065a#diff-aa3a05d9d57c08d694cf230a36386221f56890092e54cf8d6b04fab1b2cc0702) the command. Thanks

3djc, 2021-01-19:
Please confirm it's working for you (tested here, seems ok)

rotorman, 2021-01-19:
Huh, that's unfortunately too much asked - I am not yet setup to compile OpenTX on my own :(

kilrah, 2021-01-19:
You don't need to, you download the automated nightly build.

rotorman, 2021-01-20:
Thank you for your feedback!

I went ahead and installed today's nightly on my TX16S (opentx-tx16s-lua-noheli-ppmus-en-2.3.12N419.bin) and created following script that I assigned in model settings to momentary SH for test purposes:
```
local function run()
    setSerialBaudrate(57600)
	serialWrite(getValue('thr')..'\n')
end

return {run=run}
```
The good news first - I hooked the AUX1 up to a logic analyzer and observer the output - I do get nicely 57600 baud output as expected.

The bad news - after every power-on of TX16S it does not work before I do the following: in SYSTEM -> HARDWARE -> Serial port (TTL) select OFF and then select back to LUA again.

Am I doing something wrong or does the fw need a fix?

3djc, 2021-01-20:
Try with AUX2 please

rotorman, 2021-01-20:
That was a good tip - indeed with AUX2 I do NOT need to reset the Serial Port setting under SYSTEM->HARDWARE.
Thus the issue I posted 2 hours ago is only with AUX1.

But I did notice sometimes a spurious peak, please see the logic analyzer screenshot (with AUX2):
[image]

It can be seen on TX (Channel 0 of logic analyzer) and also on RX (Channel 1) lines (the 0x2D 0x32 0x0A are expected, the first peak not). Maybe it stems from the setSerialBaudrate() call.

One more thing - I tried asking RadioMaster support, but did not get an answer - is the AUX 2 CTR pin connected at all or implemented in OpenTX? And for hardware flow, CTS/RTS is typically used, but what function does "CTR" have? Thanks!

3djc, 2021-01-20:
There is definitively a very weird issue, investigating

CTR is not currently used or managed by OTX.

3djc, 2021-01-20:
@rotorman ca you test tomorow ? It should be fixed. As a side comment, you should not send you setSerialBaudrate() everytime you wrote something if you make a more complex script

rotorman, 2021-01-20:
Yes, eager to. Tomorrow I will get the next nightly and flash it and try again.

About the script, yes - this was just for testing, But I changed now also for testing it to:

```
local function init_func()
	setSerialBaudrate(57600)
end

local function run_func()
	serialWrite(getValue('thr')..'\n')
end

return { run=run_func, init=init_func }
```

rotorman, 2021-01-21:
I flashed opentx-tx16s-lua-noheli-ppmus-en-2.3.12N420.bin and can confirm that now both serial ports of TX16S (AUX and AUX2) work (also without glitches on RX and TX lines). Awesome, thank you 3djc!

My dream of being able to skip extra hardware (Mav2PT Teensy board) to get MAVLink telemetry directly into Yaapu telemetry screens has come one step closer.

I wonder what is the reason to have under SYSTEM -> HARDWARE the Max bauds setting at all? What does 400000 or 115200 change? I wonder if this setting is now necessary at all and could be removed from menu for clarity?

*****
There is presently a problem with EX1 and EX2 analog inputs, at least with nightlies N419 and N420 that I tested. When flashed back to v.2.3.11 release version, the analog input bug is away. I'll open a new issue for that.

3djc, 2021-01-21:
Max baud is unrelated to serial ports, but refers to the sport telemetry line (this affects Crossfire modules for exemple)

closed 2021-01-21 by 3djc

rotorman, 2021-01-21:
May I suggest changing the label "Max bauds" under SYSTEM -> HARDWARE to "S.Port max baud" to better differentiate it from the serial ports and to denote it's actual purpose.