opentx/opentx #8227: Companion fails to start on Fedora 33 ok on Fedora 32

MikeAndrews-NZ, 2021-01-18:
Log file is:
[companion messages.zip](https://github.com/opentx/opentx/files/5827184/companion.messages.zip)

elecpower, 2021-01-18:
A quick web search reveals a number of non-Companion crashes with Qt and Fedora 33.
Companion currently uses a very old Qt library and this looks like another case where its support/compatibility by the OSes is being dropped. Unfortunately there are so many OS variations  across Windows, Linux and Mac that the OpenTX team does not have the resources to test against and support them all.
The OpenTX devs have been reluctantly pondering, due to the work involved, the next build baseline. When released is almost guaranteed to raise new compatibility issues across the older OSes.
As the Qt issue seems to be a broader issue, keep a look out on the Fedora forums to see if Fedora releases an update/workaround to fix in the meantime.

kilrah, 2021-01-18:
Did you install the deb or build from source? Build from source should work, I have companion running with Qt 5.15.1 here.

MikeAndrews-NZ, 2021-01-18:
Thanks for the quick response.
I converted the deb to a rpm using alian. I then rebuilt the rpm using
rpmrebuild to remove the /lib entry in the %files section.If left in
the /lib entry causes an error on the install.I have been doing this
ever since you stopped building rpm. It's easy for me and I have not
seen any issues previously.
Thanks for your efforts in producing opentx - a super product.
Regards Mike

closed 2022-04-17 by stale[bot]