opentx/opentx #8234: FrSky SPI - Telemetry lost when moving sticks

AndreaCCIE, 2021-01-19:
**Describe the bug**
Hi all, I have a weird issue that I have no idea if it's up to the SPI (that has worked correctly for a while), OpenTX or the multiprotocol module on my TX16S.
It's related to the only 65mm whoop I do fly, and it's been on a shelf for a while so I can't say if it's related to a multi protocol or OpenTx specific update.

**To Reproduce**
Power up the quad, bind it
1. Move the throttle up
2. Move the pitch/roll left or right (check the video below)

**Expected behavior**
It shouldn't lose connection to the SPI

**Video**
https://www.youtube.com/watch?v=Jn5d_tS4yfw

**General information**
- OpenTX 2.9.11
- Hardware Radiomaster TX16S
- Multiprotocol 1.3.1.92

Any troubleshooting I can do to understand if it's OpenTX related?

kilrah, 2021-01-19:
FrSky protocols (which I guess you're using) are known to lose connection when the model and radio are too close, you want at least 1m or 2 between them. Maybe your tuning isn't good. But OpenTX only says "telemetry lost" if no telemetry is received from the module, so it's unlikely to be on our side.

closed 2021-01-19 by kilrah

AndreaCCIE, 2021-01-19:
You could have asked more information before closing the issue.
The tuning is good, it does happen regardless the flight controller installed, it doesn’t happen if I do not move the sticks.

kilrah, 2021-01-19:
There's nothing I could ask for since there doesn't seem to be anything that could be related to us.
I do know that FrSky compatible SPI RX has been problematic for years, there's a whole thread on RCG about it. BF has made multiple attempts at getting it to work right over the years, so have you updated BF?