opentx/opentx #8235: Font color switches from black to white randomly

b14ckyy, 2021-01-20:
Homescreen font color switches from black to white randomly. Can be triggered with LUA Widget.
This applies for multiple color screen radios

**To Reproduce**
Steps to reproduce the behavior manually:
1. Set up a new model
2. keep the first home screen page as it is (with black fonts and table frames)
3. add a second screen with the INAV LUA Telemetry Widget (no rc link or any setup needed) https://github.com/iNavFlight/LuaTelemetry/releases
4. switch back to first screen (issue might be visible already) 
5. turn off radio and turn it back on
6. > Fonts on first home screen are normal black now as they should be 
7. switch to the second screen to show the INAV LUA widget
8. switch back to first screen
9. > font color, table frames, sliders and trim position bars are all white now

A Reboot temporarily solves issue. 

If a bright background image is used, the font won't be visible anymore
Sometimes this even happens randomly without the INAV lua installed. This is just to trigger the issue manually. Also the system menu and model setup menu font can switch to white and then the menu entries are not visible anymore till they are hightlighted
This also happens with different themes so it is not Theme related. 

**Expected behavior**
Font should stay in the same color all the time. 

**Screenshots**
Will add screenshots later if needed 

**General information**
 - OpenTX 2.3.11 on Radiomaster TX16S
 - OpenTX 2.3.9 on Horus X10S without INAV LUA
 - OpenTX 2.3.9 on TX16S (need confirmation from a friend who has not updated yet)

3djc, 2021-01-20:
You have a widget that uses setColor() and does not restore it. Nothing that can be done OTX wise I'm afraid

b14ckyy, 2021-01-20:
> You have a widget that uses setColor() and does not restore it. Nothing that can be done OTX wise I'm afraid

please read again. It does also happen without the INAV LUA widget or any other widget set up. Just very rarely then. My friend with the X10S does not use the lua widget and suffers from the same issue sometimes.

kilrah, 2021-01-20:
The issue with widgets is well known, but we've never heard or seen about something like that without lua, so please detail / screenshot when that happens and provide the corresponding otx.

3djc, 2021-01-20:
Also bear in mind that a widget doesn't need to be displayed to be running. If it is defined on a non visble page, it is running

b14ckyy, 2021-01-20:
OK I think you are right guys. My friend with the horus might have been mistaken. It seems as he also used the INAV lua on his horus. So I'll open a ticket over there and see what the developer says about it.

b14ckyy, 2021-01-21:
I have opened the ticket on the Lua script git. 
One quick question before we close here. @kilrah you asked for a screenshot. How the heck can I make a screenshot on OpenTX 2.3.11. I found something with RTN+ENTER but that does not seem to work. Do I really have to sacrifice a switch for that? 
If yes I could use Throttle trim as I will never need it anyway.

kilrah, 2021-01-22:
You can trigger the "Screenshot" special function with anything you want. But taking a photo of the screen with a phone works too...

b14ckyy, 2021-01-22:
OK i was hoping for a key combination as on the X9D ^^ 
Nevermind. I'll close here. Thanks for you help guys.

closed 2021-01-22 by b14ckyy