opentx/opentx #8236: Analog input of EX1 and EX2 on TX16S not working correctly, at least with v2.3.12 nightlies 419 & 420

rotorman, 2021-01-21:
**Describe the bug**

I have analog pots connected to SI & SJ analog pins in TX16S:
[image]

When testing UART with v2.3.12 nightlies N419 and N420, I noticed a problem in both nightlies with outputting these channels as EX1 and EX2. The range is very limited to only one end of the pot where the values go from 0 to 100% quickly. Also the output is not going to other direction. It seems as if the gain factor and offset of EX1 and EX2 mapping might be off, as in the SYSTEM->HARDWARE -> Debug -> Analogs I can see that both pots are sampled correctly from 0x0000 to 0x0FFF.

I flashed back to v.2.3.11 release version, and there it works again OK, so I'm pretty sure it is not a hardware issue on my side.

**To Reproduce**
1. Connect 2 pots on TX16S to SI & SJ analog ports
2. Under SYSTEM -> HARDWARE -> Pots, enable EX1 and EX2, in my case "Pot with detent"
3. In model setup under inputs, I enabled 05 as EX1 and 06 as EX2 with default settings
4. In model setup under mixes, enable CH5 as I05 and CH6 as I06 with default settings
5. Observe the input in Channels Monitor 1-8 "Monitors" screen while turning the pots.

**Expected behavior**
Full range from -100 to +100% (988us to 2012us), as with v2.3.11 release version.

A clear and concise description of what you expected to happen.

**General information**
 - OpenTX version: N419, N420 have the bug, release v2.3.11 not
 - Hardware: RadioMaster TX16S with custom pots on SI & SJ analog inputs

3djc, 2021-01-21:
We havent changed anything there, is you calibration ok ?

rotorman, 2021-01-21:
I did recalibrate with N419 and N420, and turned the pots during calibration from one end to other (there is no feedback in calibration screen though, but turned the pots anyhow). Also when first asked for them to be center, I made sure my extra pots were also centered.

3djc, 2021-01-21:
I have made a version that allows you to see those two when calibrating. Could you try and let us know how it behaves at calib ? (the pots you have a really with detent, right ?)
[firmware.zip](https://github.com/opentx/opentx/files/5848104/firmware.zip)

rotorman, 2021-01-21:
Awesome, thanks - I flashed this FW to my radio and now I can nicely see also the EX1 and EX2 pots during calibration. Here the pots move perfectly from one side to other. Now also the Channels Monitor screen works correctly (and I can assure you I did move during the calibration of N419 and N420 also them from one end to other, this worked with v2.3.11 release flawlessly, but did not in N419, N420).

BUT, possibly the Preflight check for EX1 and EX2 needs a fix. Here some very weird value is used as a center (in comparison with v2.3.10 and v2.3.11 release, where they workes as expected).

To answer your question - the pots do not really have a detent (they are [Jeti Stick pots](http://www.jetimodel.com/en/katalog/Transmitters/Transmitter-Accessories/@produkt/Stick-for-DC-16-Potentiometer/), I'll upload some pictures of this modification in RCgroups soon), but I have to use "Pot with detent" setting if I want them to be included in the Preflight Checks checked for *centering* on start-up. If I use just "Pot", then one end of the stick is checked and not the center.

Maybe I do not fully understand the meaning behind Preflight Checks -> Pots & Sliders -> Manual versus Auto? I tried both settings and the preflight check with the firmware you sent me, works only in manual mode, in auto mode their state seems to be always ignored.

3djc, 2021-01-21:
Ok, now I'm totally lost, since the ONLY change in that firmware is displaying the values, not treatment whatsoever (https://github.com/opentx/opentx/pull/8237)

If those pots don't have detent, then do not set them as pots with detent, but as regular pots (so POT)

For checks, auto means it remembers how the where when you last finished using the model
In Man mode, select pos position line, click enter and select a particular pot. Enter wil enable or disable the check, and long enter will record the position to be matched (there is a bip sound to indicate the position has been recorded)

wimalopaan, 2021-01-21:
Sorry for hijacking: how did you fit the JETI Stick pots into the TX16s?

3djc, 2021-01-21:
He said just above that he was going to post it on RCG ....

rotorman, 2021-01-21:
My bad - I finally now understood how the Auto/Man functions - the Flight Check works as expected, I also here a short beep for saving the centers with long press - all good, and sorry for the extra trouble.

It would be awesome if you could merge the calibration update code, that is very useful for EX1 and EX2 And again a big thank you!

@wimalopaan I will post here the link to RCGroups post when I have done it (I expect to do it today).

3djc, 2021-01-21:
Thanks for the report, closing it then :). The calib will be merged next round of peer review

closed 2021-01-21 by 3djc

rotorman, 2021-01-21:
@wimalopaan You will find more info about the installation in a RCGroups post here:
[https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page603#post46367179](https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page603#post46367179)

rotorman, 2021-01-21:
As I struggled intuitively understanding the Manual vs. Auto under Preflight Checks -> Pots & Sliders, may I suggest: "Preset" and "LastPos" as alternative menu titles for "Manual" and "Auto". IMHO these would be more intuitive to understand. Just a tip. Thanks!

kilrah, 2021-01-22:
Not going to change something that people know and has been documented for years...

rotorman, 2021-01-22:
ok!