opentx/opentx #8241: Anomaly in the FRSKY telemetry output stream on TX16S

b14ckyy, 2021-01-23:
There is some kind of anomaly in the Telemetry output of the Radiomaster TX16S that makes at least one known device as a receiver unusable. This affects a aftermarket installed genuine Horus X10S BT 

Setup: 

- Radiomaster TX16S MAX
- Upgraded with Horus X10S Bluetooth LE Module
- Internal R9ACCESSMOD installed
- Genuine DSD Tech HM-10 BTLE Module paired with the radio BT
- OpenTX 2.3.11 with Bluetooth and INTERNAL R9 ACCESS Mod enabled (also tried 2.3.12NB421 with same issue)
- Antenna Tracker with U360GTS Universal Telemetry Controller

Before the TX16S I used a Taranis X9D with an TTL Inverter and a Pair of HC-05 BT modules on 57k baud to mirror telemetry to the tracker and that worked fine with 2.4GHz and R9 System. 

Now after switching to the TX16S the same chain was built just with HM-10 and Horus BT module. But now the following issue appears: The U360GTS Protocol detection, recognizes a FRSKY_X Telemetry stream but does not show any data. All packages are disregarded as corrupt. I have tested many configurations and it was impossibly to get the tracker reading the telemetry from the radio. The Android FrSky Telemetry Viewer app on the other hand works fine in multiple chains (see test setups below). But the telemetry transmission from a Crossfire Module over BT works just fine and correctly. 

Steps to reproduce the behavior:

1. Take any TX16S (maybe BT mod is not even needed, Telemetry mirror to UART works too)
2. Bind a plane with FrSky 2.4GHz D16 receiver or use R9 System
3. Enable telemetry mirror on UART or BT
4. Flash U360GTS firmwareTo a NAZE32 or SPRF3 Board https://github.com/raul-ortega/u360gts-configurator
5. Connect OLED screen over SPI and enable Protocol auto detect and provider INAV (optional, add buttons to use the OLED menu)
6. Bind HM-10 to the Horus module and connect it to the tracker
7. Power all up .... 
Now the Tracker will detect FRSKY_X Protocol but all data stay 000 

I have tested all the following link chains to rule out as much as possible: 

##################
I always used a plane with either R-XSR receiver (Multi Module) or R9mm receiver. Both always show same result.
X9D - Serial Out - HC-05 baud 57k ..... HC-05 57k - tracker               | fine
TX16S - internal BT on Mirror mode ...... HM 10 115k/57k - tracker     | problem
TX16S - HM10 on to keep Horus BT active - HC-05 115k connected to UART2 tx (grabs date before the BT module) ..... HC-05 115k - tracker     | problem
TX16S - Android Phone connected to internal BT - Telemetry viewer App       | fine
TX16S - HM-10 115k connected to TTL-USB Adapter - TTL Adapter on phone via USB - Telemetry Viewer app | fine

Crossfire Full with Micro Rx - TX16S - HM-10 - tracker                  | fine  _

**Expected behavior**
After testing all these chains in comparison, I can rule out problems with the BT connection, Horus module, Tracker and the hardware in general. Basically the Telemetry transmission to external devices works as the Telemetry Viewer App had no issue in all constellations. But something in the Data stream must be different to the Taranis X9D with the same OpenTX version because the tracker firmware has no problem to read the data there, but does not work with exactly the same transmission chain (2x HC-05). The only difference here is the radio model. 

**Screenshots**
[image]

**General information**
 - OpenTX version 2.3.9+2.3.11 on X9D (working); 2.3.11+2.3.12NB421 (not working)
 - Hardware Radiomaster TX16S with internal Horus BT Mod
 - OpenTX Companion 2.3.11

3djc, 2021-01-24:
If you have the R9M access mod, are you then using a 2019 R9M in access mode ? Because we already have an issue open that some trackers have an issue with that

b14ckyy, 2021-01-24:
> If you have the R9M access mod, are you then using a 2019 R9M in access mode ? Because we already have an issue open that some trackers have an issue with that

Yes I use it in access mode. But the issue also appears with the R-XSR receiver through the internal multi module and no R9 module installed.

A friend of mine who also participated in the U360GTS development sent me a test firmware he made a while ago where he disabled the CRC checks from frsky telemetry. He did that as a workaround a while ago because some radios did not work properly. With the disabled crc checks I now see coordinates. 

That means the telemetry stream from the TX16S seem to not have valid crc checks in the stream. But from the X9D it has. This is only a workaround as I sometimes get corrupted packeges and wild jumps in altitude or position randomly. I can see the same glitches on the telemetry viewer app but only with the TX16S.

3djc, 2021-01-24:
Cannot be CRC treatment as their is a single (same) piece of code for all radios

b14ckyy, 2021-01-24:
> Cannot be CRC treatment as their is a single (same) piece of code for all radios

Then maybe a high data error rate in general? I just can tell you what I'm seeing. With the workaround firmware on the tracker I suddenly see the GPS data on the tracker correctly (beside the position and altitude glitches when it suddenly has false data for a second). 
Baud rate to the internal Horus BT Module be the problem to be too high? As I see on the UART2 TX pin (where i also can see the AT commands sent to the BT module) it has a 115200 baud. On the other hand the X9D works with a 57600 baud on the telemetry mirror port. 

Ii could flash back to a non-access firmware for testing to get the UART1 on the radio back. What is the baud rate these UARTs run at in telemetry mirror mode? Is it also 115200?

3djc, 2021-01-24:
The Horus BT does have a custom firmware we do not not know anything about, beside the fact that several, non compatible, versions seem to exist

b14ckyy, 2021-01-24:
> The Horus BT does have a custom firmware we do not not know anything about, beside the fact that several, non compatible, versions seem to exist

I know. But I have ruled out the Horus module already by grabbing the Telemetry data from UART2 Tx pin, before they reach the BT module. And then used the HC-05 to send the data from there to the tracker. With the exact same result. So the issue in the stream already occurs before the Horus BT module.

3djc, 2021-01-24:
Well, what you see there is data formated as per required for the FrSky BT module requirement, I would not expect it to work for HC-05

b14ckyy, 2021-01-24:
the data stream is just pure telemetry sent to the BT modules uart. When I connect the second HC-05 to my phone with a TTL-USB adapter I can see the telemetry ont he telemetry viewer app. So the telemetry data are there and transmitted. 

Another thing I just test right now: 
Direct wire from UART2 Tx pin (grabs data before the Hous module) to the tracker with CRC disabled firmware: Works and I see data. 

I can 100% rule out the BT as root cause.

b14ckyy, 2021-01-24:
> Well, what you see there is data formated as per required for the FrSky BT module requirement, I would not expect it to work for HC-05. You could test telem mirror on the other port

Do I need an inverter to use the UARTs in native telemetry mirror mode. I have flashed a non-access non BT firmware before to get back both UARTs one in the OpenTX settings and I tried but I did not get any data at all sometimes and even if I get data out, even the telemetry viewer app refused to work. 

what is the BAUD rate of these UARTs in telem mirror mode?

kilrah, 2021-01-24:
The other issue lists a difference between modules that send extra 0x7Es between frames and it seems that particular project doesn't properly ignore them.

b14ckyy, 2021-01-24:
> The other issue lists a difference between modules that send extra 0x7Es between frames and it seems that particular project might not properly ignore them.

But as written above, I have completely ruled out the BT module by grabbing the data stream before the module and direct wired it to the tracker. Same result. I see the GPS data (with some corrupted packages) on the tracker with CRC check disabled. And with no data at all with CRC check enabled. So Bluetooth module can be ruled out completely.

b14ckyy, 2021-01-24:
Do I need an inverter when I use telemetry mirror on the UARTs? Because I tried that before and was not able to get any telemetry out from there. And Baud is also 115k or 57k like on the X9D?

kilrah, 2021-01-24:
> But as written above, I have completely ruled out the BT module 

Yes I meant RF module, not BT module.

b14ckyy, 2021-01-24:
> > But as written above, I have completely ruled out the BT module
> 
> Yes I meant RF module, not BT module.
> Inverter, well it depends on what you're connecting to it.

But that would be a very big coincidence if the internal 4in1 multi module and also the R9M would do the same. 
OK I will flash non-access firmware again and then test the UART1 Telem mirror again with the internal Muli and the R-XSR and hope I don't need an inverter. Otherwise I have to solder out the one from the X9D as it's the only one I have.

b14ckyy, 2021-01-24:
> > But as written above, I have completely ruled out the BT module
> 
> Yes I meant RF module, not BT module.
> Inverter, well it depends on what you're connecting to it.

I just tried again to get telemetry via UART by flashing a non-ACCESS firmware to get back UART1. I used the Internal Multi Module and removed the R9 Module. I get data output when I read it in the serial monitor on PC. But by no chance does the Telemetry viewer app get any valid data, nor the tracker. With an inverter I get no data at all. I tried with HC-05 to phone, TTL-USB Adapter to phone and Directly wired to the tracker. Nothing....

b14ckyy, 2021-01-26:
Just yesterday I was pointed at the same issue int he MFD Crossbow group with the Horus X10. Here is the post: 
[image]

So the telemetry data are fine when they come from the module but don't work when they output over the BT module. I can't test the output in my module directly as I'm on access and there the S.Port is disabled.

b14ckyy, 2021-01-27:
Update: 
I got 2 test Firmware for the U360GTS. There are changes to address FrSky telemetry issues from Horus Style radios that popped up already in summer 2020. 

One with CRC Checks completely removed when Frsky_X telemetry comes in. 
**Result:** Now I get GPS Coordinates, Speed and Altitude on the tracker but at a very low update rate of under 1Hz. It basically updates only occasionally and sometimes the data glitch and the GPS coordinates or altitude are off by a huge amount. Basically the plane virtually jumps to other hemispheres. I have seen a similar problem with the Telemetry viewer app before. 

Another one with a newer and better CRC algorithm. 
**Result:** This one seems to work perfectly fine. Update rate was at stable 4Hz with no data glitches. No outdoor test yet but it looks promising. 

But this won't solve the Telemetry viewer App telemetry glitches and also not the much more used Crossbow Tracker issues. Basically these patches are just a workaround. I never saw these kind of issues with a Taranis radio and Bluetooth or wired use and I always had a stable 2-3Hz Update rate there (lower baud rate). 

So the root cause still must be on the radio/openTX side, causing strongly corrupted telemetry data.

ampmam, 2021-01-28:
Not sure if my issue is the same as yours but my issue is as follows:
1. I am on 2.3.11
2. Bluetooth HC05 is configured for the baud rate 115200 using Arduino
3. Bluetooth HC05 is connected on UART1. Enabled HARDWARE>Max bauds 115200, Serial port (TTL) Telem Mirror
4. When I use R9M, I get full telemetry data on the Telemetry Viewer app but when I use Internal RF Multi Frsky D16, I do not get any telemetry data on the Telemetry viewer app. I tried changing the baud rate from step 2 to 57600 but no luck.

3djc, 2021-01-28:
@ampmam, perfectly normal, with R9M you get FrSky telem, with multi you get Multi telem. I guess you telem app doesn't know how to decode multi telem

ampmam, 2021-01-28:
@3djc I heard some people had success getting multi telem to the same "Telemetry Viewer" app some time back.  BTW, the app uses LTM protocol and I believe it should be able to decode it. I also tried in few other android apps i.e. EZ Gui Ground station with no luck, I get the message "Data not received". I feel for some reason, using the Multi Telem, UART is not passing the data through. It may just be my case but I like to hear from other people too.

3djc, 2021-01-28:
Seems ok to me

[image]

kilrah, 2021-01-28:
When using multi the telemetry of the different protocols is wrapped into multi telem packets. If the app doesn't know how to strip the wrapper it won't work

ampmam, 2021-01-28:
@kilrah If that is the case probably the MPM dev @pascallanger can add the logic not to wrap it into multi telem packets if the selected protocol is Frsky but I completely understand it may be too much to ask. IMO, Frsky telem would be more useful than multi packets telem when using Frsky protocol. Just my two cents.

pascallanger, 2021-01-28:
Nope, there is more than just the FrSky telemetry in the multi telem packets... I won't touch it. Write a LUA script to rewrap it if you need to.

b14ckyy, 2021-01-28:
I had no luck with uart telemetry at all. But with the internal BT mod telemetry both, the R9M and also the internal Module output FrSky telemetry. Works witht he telemetry viewer app on both.

ampmam, 2021-01-29:
@b14ckyy Good to know the internal BT works on both. Are you using the Horus BT or any other kind? If the internal BT works, it may be the perfect solution to use on Jumper T18 with both D16 and R9 protocol.

b14ckyy, 2021-01-29:
@ampmam I use the Hourus Module. The BT connection has not the best range but it is stable for at least 5m in my flat. Maybe more outdoors. 
But keep in mind there is still the initial issue I opened this ticket with. Neither with the BT nor with the grabbed telemetry on UART2 before the bt module, the MFD Crossbow will work. For the U360GTS there might come a workaround firmware soon but the Telemetry Viewer App position glitches are still an issue.

raul-ortega, 2021-04-07:
Dear @b14ckyy current u360gts development version is working with radiomaster TX16s + Horus Bluetooth module. It has a some contribution that reworked the CRC calculation for FrSky X protocol and it seems to work.

I say 'seems' because the u360gts understands frsky x telemetry sent from opentx (and also Telemetry Viewer app for android). However, I have detected some failures during tracking that might be related with "AT+TXPW0" at command sent from opentx to configure the power of the bluetooth module. This might be causing some signal loss betewwn the Horus Bluetooth module and the HM-10 bluetooth module installed inside the tracker.

b14ckyy, 2021-04-08:
Hey @raul-ortega , ist that similar to the changes that Albert made last year and not implemented into master? If yes then it will work for sure as he made me a build with these changes that works flawlessly. I won't e able to test the tracker again till may/june unfortunately.

raul-ortega, 2021-04-08:
Dear @b14ckyy, is more than probably that for some mistake that commit was
not included in the .hex file published as v v11.1.0
<https://github.com/raul-ortega/u360gts/releases/tag/11.1.0> release. That
pull request was accepted and included, development branch is up to date
and working with my radiomaster + BLE + R9M . Not only for me, one friend
of mine has tested it too and also works. I have also include the feature
to use the tracker as a transparent bridge (telemetry fordwarding) and as
soon as I have some free time I'll publish a new release (configurator also
needs to be updated).

Despite of that, telemetry over bluetooth glitches as happening

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

raul-ortega, 2021-04-08:
https://www.youtube.com/watch?v=HTbRv6JHI98

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2022-04-16 by stale[bot]