opentx/opentx #8250: CCM and RAM usage: ????

olliw42, 2021-01-26:
This is not a bug report in the sense that something would not currently work, yet a report on what I think seems to be an unfortunate use of the F4's read-write memory areas.

Since starting to add code to opentx radio I repeatedly had the feeling that the read-write memory is nearly exhausted, although it wouldn't be clear to me for what. I now worked on adding a further task and ran into the issue again, so I investigated more carefully, and noted the below, which allowed to solve the issue for me.

To start with I should say that I'm really not an expert on linker scripts, so my apologies.

My conclusion in short: 

I believe that of the two read-write memory areas which the F4 provides, CCM and RAM, only the CCM is used while the RAM is massively underused.

**Details:**
---

In the current linker script for the horus, https://github.com/opentx/opentx/blob/2.3/radio/src/targets/horus/stm32f4_flash.ld, one can find this (where I only kept the relevant parts):

```
_Min_Heap_Size = 0;      /* required amount of heap  */
_Main_Stack_Size = 1024; /* required amount of stack for interrupt stack (Main stack) */
...

MEMORY
{
  ...
  RAM (xrw)       : ORIGIN = 0x20000000, LENGTH = 192K
  CCM (xrw)       : ORIGIN = 0x10000000, LENGTH = 64K
  ...
}

SECTIONS
{
  ...

  .data : AT ( _sidata )
  {
  ...
  } >CCM

  .bss :
  {
  ...
  } >CCM

  .noinit (NOLOAD) :
  {
  ...
  } >CCM

  .ram (NOLOAD) :
  {
  ...
  } >RAM

  ...
}
```

When I'm compiling my current branch (with a modified ld script, see below), and am using `COMMAND arm-none-eabi-size --format=sysv firmware.elf` to display the sizes of the .data, .bss, and .ram sections, it yields

.data = 880
.bss = 71096
.ram = 16580

If I'm compiling with the linker script as is, i.e., with .data > CCM and .bss > CCM, when I get a linker error saying `region CCM overflowed by 6440 bytes`.

Now, that makes sense. If both .data and .bss are located in CCM, which has a size of 64K, one finds that 880 + 71096 - 65536 = 6440, which I take as evidence that my thoughts are not nonsense.

So, I tried other combinations. Here the results:

- **.data > CCM, .bss > CCM:**  overflow by 6440 bytes, check: 880 + 71096 - 65536 = 6440

- **.data > RAM, .bss > CCM:** compiles and links fine, and according to bench testing works fine

- **.data > CCM, .bss > RAM:**  overflow by 5560 bytes, check: 71096 - 65536 = 5560

- **.data > RAM, .bss > RAM:** compiles and links fine, and according to bench testing works fine

**Conclusions**
---

So, my conclusion is, that currently essentially only the stack is placed into RAM, but both .data and .bss go into CCM. 

One can gain more memory accessible to the code by putting the .bss section, since its so big, into RAM.

I wonder if it could be reasonable performance-wise to put the stack into CCM. 

**Unknowns to me**
---

At least two points are unclear to me. First, I'm not sure how the lua memory is handled in the code. There is no heap, so it's not using this directly, but I could imagine that you somehow got it placed into what is left over in RAM. If that were true the above change would reduce the lua ram. Secondly, it is not clear to me where the Cocox RTOS is placing its key variables and task stacks. If they are placed in .bss, then moving .bss to RAM could technically mean less performance.

I should say that I'm using a reasonably sized script, and didn't encounter any issues (yet?) with the two options which make it work for me.

bsongis, 2021-01-26:
Malloc is done in RAM and Lua needs as much RAM as possible.
If we reduce RAM space available, we will get a ton of user reports that their script doesn't start anymore!

closed 2021-01-26 by bsongis

olliw42, 2021-01-26:
thx for the clarification!

so, this is then a very severe limitation on what is possible ... I was surprised to see how much ram a cocox task is eating up ... well ...

THX