opentx/opentx #8252: Companion overwrites timer setting.

RC-SOAR, 2021-01-26:
In the Model Setup menu on the transmitter, there is a setting for the start of voice/beep countdown. However, this is missing in Companion (see screenshots). 

Also, if models and settings are saved using Companion, then written back to the transmitter, the setting is overwritten with a default value of 10. Steps to reproduce:

1. On the transmitter, create a new model
2. In the model setup menu, set a timer with beep, and start of countdown = 20
3. Using Companion, save models and radio settings, then write back to transmitter. The start of countdown is 10 instead of 20.

Setup
- OpenTX 2.3.11
- Horus X12S (also applies to other tx's)

[image]

[image]

kilrah, 2021-01-26:
I wonder if anyone has ever used that feature since it's been like this since 2016, and the option has been missing on 128px radios since then as well...

RC-SOAR, 2021-01-26:
It's a useful configuration option for sailplane flyers. The alternative is to read out timer values via special functions, but each value is accompanied by 'seconds' and it doesn't work well.

3djc, 2021-01-26:
We agree, we where just shocked that this didn't get noticed earlier. Complete fix is on its way

RC-SOAR, 2021-01-26:
Great - much appreciated as always :-)

PR #8253 (2021-02-02): Timer countdown start missing in companion and 128 (fix #8252)
This fixes #8252
commit ffa6a75b (2021-02-02): Merge pull request #8253 from opentx/8252_countdownStart

Timer countdown start missing in companion and 128 (fix #8252)
closed 2021-02-02 by bsongis