opentx/opentx #8257: Telemetry simulator should provide VFR telemetry for FrSky systems at least

pr-newman, 2021-01-27:
Desire: 
It would make Telemetry Simulator more useful if it provided VFR telemetry for FrSky systems.  

Problem:
The change for FrSky systems from OTX v2.3.8 onwards replacing FLR (Frames Lost Rate) with VFR (Valid Frames Rate) required a change to a lot of model setups to replace FLR with VFR in Logical Switches etc after updating the Tx firmware (I am using v2.3.11).  

The absence of FLR from the Simulator telemetry was previously not noticeable, because FLR=Absent was equivalent to the benign FLR=0 (ie frame reception is good).  However, the absence of VFR from the telemetry is not benign as it now looks like the valid frame rate is zero (ie bad!).  

Aside from needing to change the LSs and GVs testing FLR in all my models to test its inverse VFR, I am unable to sensible use the Simulator to test the model setups without my LSs continuously asserting that VFR=0 which is an undesired flight condition.

I checked as far as I could to see if this was already reported and did not find it as a bug or an undesired feature but I recognise this would require an addition to the Telemetry Simulator code rather than a correction.

lshems, 2021-01-27:
Couldn't you write a Lua script faking a telemetry sensor? Example on request

pr-newman, 2021-01-27:
Hi Ishems thanks for a quick response!  If I knew LUA code then maybe I could do as you suggest but sorry to say I don't.  It would also mean (another) change to all my model setups I think?As I said in the request though, the change of meaning from FLR to VFR makes the presence of a non-zero/non-absent value for VFR required to indicate a good flight condition.  So I think including it in the Telemetry Simulator should have been part of the same change that inverted FLR.Cheers --Pete

  for FrSky systems at least (#8257)

lshems, 2021-01-27:
Ok.

closed 2022-04-17 by stale[bot]