opentx/opentx #8260: Taranis X-Lite trim switch does not match stick mode (aileron on left/right stick)

123jhe, 2021-01-29:
On Taranis X-Lite Pro the trim switch cross operates with a shift key:
- trim right/left without shift trims aileron
- trim right/left with shift trims rudder

The idea is that trim switch works for **right stick without shift** and for **left stick with shift** key. But this configuration only works for mode 2 stick configuration. For mode 4 (aileron on the left stick), it creates a problem, since the up/down trim without shift takes effect on the right stick and left/right trim on left stick. This is inconsistent behaviour.

**To Reproduce**
Steps to reproduce the behavior:
1. set transmitter stick mode to 'mode 2' (aileron on **right** stick)
2. try left/right trim --> it takes effect on **right** stick
3. try up/down trim --> it takes effect on **right** stick
4. set transmitter stick mode to 'mode 4' (aileron on **left** stick)
5. try left/right trim --> it takes effect on **left** stick
6. try up/down trim --> it takes effect on **right** stick

**Expected behavior**
Trim switch mapping should follow the stick mode (1 to 4) configuration: 
- Trim takes effect on functions configured on right stick if shift key NOT pressed. 
- Trim takes effect on functions configured on left stick if shift key pressed.
- Same principle applies to throttle/elevator mapping as well.

Steps to test the correct behavior:
1. set transmitter stick mode to 'mode 2' (aileron on **right** stick)
2. try left/right trim --> it takes effect on **right** stick
3. try up/down trim --> it takes effect on **right** stick
4. set transmitter stick mode to 'mode 4' (aileron on **left** stick)
5. try left/right trim --> it takes effect on **RIGHT** stick
6. try left/right trim **with shift** key --> it takes effect on **LEFT** stick
7. try up/down trim --> it takes effect on **right** stick

**Screenshots**
[image]

**General information**
 - OpenTX version : 2.3.11
 - Hardware : Frsky Taranis X-Lite Pro
 - OpenTX Companion : 2.3.11

kilrah, 2021-01-29:
> **To Reproduce**
> Steps to reproduce the behavior:
>
> 1. set transmitter stick mode to 'mode 2' (aileron on **right** stick)
> 2. try left/right trim --> it takes effect on **right** stick
> 3. try up/down [...]

That's exactly as intended, it does follow the mode setting so that the "unshifted" trim keys always trim aileron end elevator.

closed 2021-01-29 by kilrah

123jhe, 2021-01-29:
Thanks for the feedback. I understand what you say. But how can that be intended?

In my opinion this is an issue. Why? Compare this behaviour with standard transmitters: You have two standard trims on each stick. If you change from mode 2 to mode 4, the aileron changes from right to left stick, as do the trims. If the trims would behave like in the X-Lite, the trim switch for aileron would stay on the right side, although aileron has changed position. Would make no sense.

If this is no issue: Do you think I should post it as a feature request?

Thank you

3djc, 2021-01-29:
No, it was purposefully done this way, because this way, you always have the most critical trims (ail and elev) as non shifted

kilrah, 2021-01-29:
> Thanks for the feedback. I understand what you say. But how can that be intended?
>
> In my opinion this is an issue. Why? Compare this behaviour with standard transmitters: You have two standard trims [...]

Well this is not a standard transmitter with 2 trims on each stick, so it makes no sense to treat it as one :)
Given the available 2 directly accessible trims what makes most sense is to have them control the most important functions by default.