opentx/opentx #8262: opentx-2.3.11 and windows 10 USB joystick unable to calibrate both axis.

propagandazoo, 2021-01-29:
Since updating Taranis XD9 to latest firmware opentx-2.3.11 I can no longer calibrate both axis.

I believe it should now be class compliant using windows game controller settings to solve the previous problem of driver update disruption described here: 

https://www.rcgroups.com/forums/showpost.php?p=42723295&postcount=54967 )  

Described on opentx-2.3.11 download page: 

"A while ago, Microsoft / ST.com introduced though Windows update a change to the handling of their driver which affected OpenTX users trying to use their radio a USB joystick. Based on a suggestion from Joel Stewart here, we have applied for, and gained access to, a unique ‘usb id’, so starting with 2.3.11, OpenTX users won’t be affected by this issue anymore." 

However, the driver does not seem to be working properly.

3djc, 2021-01-29:
Working fine here, maybe some leftover of your previous driver ?

propagandazoo, 2021-01-29:
> Working fine here, maybe some leftover of your previous driver/joystick settings ?

3djc are you using an XD9? 

If so it is not so good for me. I already uninstalled the previous driver.

closed 2022-04-17 by stale[bot]