opentx/opentx #8263: IRX4 won't boot unless model switched

cru774, 2021-01-30:
Created a new model in TX that uses external IRX4 module. Turn off TX, Turn on TX, module will not boot up. Select a different model, switch / select back - module boots up.

**Expected behavior**
Module boots up every time the TX is turned on.

**Screenshots**
[image]

**General information**
IRX4 Lite firmware v1.3.2.12
OpenTX v2.3.11

cru774, 2021-02-06:
By coincidence I narrowed the issue down. What you see in the video here - https://youtu.be/A7X70Z92Wy4 - is that by quickly clicking away the warnings (which I usually do), you can prevent the module from coming online entirely. If you select another model, switch back, the module comes online. Interestingly, if you wait a few seconds before clicking away the warnings, the module comes online on first try.

Not sure if someone is following this? Did I miss to provide required information?

Thanks a lot
Christian

benlye, 2021-02-06:
This is a known problem with the original bootloader on the IRX Lite module.

Have you updated the bootloader?

cru774, 2021-02-06:
Hi benlye, thank you for your reply. Honest answer: Yes, but I am unsure if it was successful. I am able to flash the module from the transmitter. I did flash with flash-multi, and the device shows up as DFU-device ever since. Is there a definitive way to tell, wether or not the boot-loader has been flashed?

I also followed all instructions on https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md#download-the-multiprotocol-source-and-open-the-project with an FTDI adapter. That also did not succeed, as when uploading from Arduino IDE, the log stated that no module was found in Boot0 mode on COM6 (which was the port the adapter was connected to). 

So I not sure how to verify...

benlye, 2021-02-06:
You can use Flash Multi to update the bootloader:
https://github.com/benlye/flash-multi#upgrade-module-bootloader

You need to write the firmware again after flashing the bootloader updater.

cru774, 2021-02-06:
I did try flash-multi a few days ago, also attempting to flash the boot-loader. This is what is happening and where I am stuck:

[image]
[image]
[image]
[image]
[image]

cru774, 2021-02-06:
Installing the USB-drivers from the flash-multi menu also fails...
[image]

benlye, 2021-02-06:
Most likely the module's firmware is too new (does not have USB support) so the app is struggling because the matching bootloader is not installed.

Can you use your radio to downgrade the module firmware to v1.3.1.59, then try to update the bootloader again?

The driver installation is a puzzle - what does the verbose output show?  Sometime the installer files get quarantined by antivirus software.  You may need to get that working, but first try without and see if the devices show up correctly when running the older firmware..

benlye, 2021-02-06:
Also, if flashing / reading continues to fail please post the verbose logs.  

And lastly, make sure you have extracted *all* the file from the zip file, including the folder structure.  I'm wondering if the app can't find the tools where it expects them.

cru774, 2021-02-06:
Thanks for your help, really appreciated. 

I did uninstall all drivers for this device, and ran flash-multi as admin. Installation of drivers succeeded, and I am now able to run "Flash Module Bootloader" from the menu. It succeeded, and I followed the instructions from the dialogue:

[image]

I went ahead and downgraded to 1.3.1.59 successfully via transmitter. "Read Module" now works:
Reading MULTI-Module via native USB
Reading flash memory ... done

Multi Firmware Version:   1.3.1.59 (STM32F1)
Firmware Target:          STM32F103CB (128KB)
Expected Channel Order:   TAER
Telemetry Type:           OpenTX or erSkyTx (MULTI_TELEMETRY)
Invert Telemetry Enabled: True
Flash from Radio Enabled: True
Bootloader Enabled:       True
USB Serial Support:       True
Serial Debug Enabled:     False

EEPROM Global ID:         0x667A9B13

MULTI-Module read successfully

benlye, 2021-02-07:
OK, so now you have the latest bootloader installed you can update to the latest module firmware again and see if your original issue has gone away or remains...

cru774, 2021-02-07:
So I did, and the process succeeded. However, the issue form my 2nd post still persists.

[image]

benlye, 2021-02-07:
> So I did, and the process succeeded. However, the issue form my 2nd post still persists.

Well that's unfortunate.  At least now we know that the problem isn't the old broken bootloader (which really does cause this issue on these modules).

cru774, 2021-02-07:
Just to reiterate: The remaining issue is, that if I click away the throttle warning / switch warning too early, the module does not boot entirely, and goes offline.

If I am waiting a few seconds, then click it away, the module completes the boot-up. 

This is some weird race condition in conjunction with speech output or user dialogue handling in OpenTx, if I would have to make a very uneducated guess ;)

cru774, 2021-02-07:
Even with the module booted up (green and red LED on constantly), I get no info in the Tx about it
[image]

closed 2022-04-17 by stale[bot]