opentx/opentx #8270: Cannot bind R-XSR with QX7

Sebitalugano, 2021-02-05:
My taranis qx7 worked with an R-xsr receiver but after updating the internal module and the receiver I couldn't bind it anymore.
I returned the firmware to the previous versions and the only thing I get is that the red led blinks at the moment of binding and when turning on again the green that confirms the link is not fixed.
I took it to a technician but he says that everything is ok, it must be a software issue but he does not know the issue.
Has anyone had this problem and could solve it? I went through all of Youtube and found nothing.

This is my own video:
https://youtu.be/MQPIrBOQ7gs

kilrah, 2021-02-05:
Binding and RF transmission issues are unrelated to OpenTX, they're pretty much always mismatches of RF module and RX firmware versions and variants.

closed 2021-02-05 by kilrah

Sebitalugano, 2021-02-10:
Ok, can you send me the files please?

I try all posibilities downloaded from official pages, and don't have results.

Taranis QX7 & R-XSR

Hope you can help me, thank You

Cheers

Obtener Outlook para Android<https://aka.ms/ghei36>