opentx/opentx #8271: Companion: Telemetry not available as a Source

sande005, 2021-02-05:
OpenTX Companion 2.3.11
Radio Profile for Taranis Plus

Telemetry entries no long show as sources in Companion.
They do appear as selections in the radio.
See this link for screen shots comparing older version to current:

[https://www.rcgroups.com/forums/showpost.php?p=46474969&postcount=4419](url)

Miami-Mike, 2021-02-05:
Which Companion versions have this problem? It's confirmed to be in 2.3.10 and 2.3.11, and not in 2.2.4, but I need to know the range of versions that are affected so that I can write an RCGroups blog post about how it relates to setup of my [LiPo Fuel Gauge](https://www.rcgroups.com/forums/showthread.php?2993146-Taranis-LiPo-Fuel-Gauge%21).

Thank you,
Mike

elecpower, 2021-02-05:
@Miami-Mike working in my copy of 2.3.10 but is broken in 2.3.11 and 2.3.12 nightlies

Miami-Mike, 2021-02-06:
Thanks. Barry Cole [posted on RCGroups](https://www.rcgroups.com/forums/showthread.php?p=46470881&postcount=4409) that it wasn't working in 2.3.10, so [I just asked him if he was sure about that](https://www.rcgroups.com/forums/showthread.php?p=46481295&postcount=4428).

Miami-Mike, 2021-02-07:
Judging by reports on [RCGroups](https://www.rcgroups.com/forums/showthread.php?p=46486657&postcount=4435) and experiments that I did, it seems that if you have an otx file that was created with an earlier version of Companion 2.3, containing a model that has a telemetry sensor set up as an input, and you load that otx file into 2.3.10 or 2.3.11, that input will be broken; the Source field will show as empty. Apparently with Companion 2.3.10 you can repair it but with Companion 2.3.11 you can't.

PR #8272 (2021-02-11): Add telemetry sources for inputs
Fixes #8271
closed 2021-02-11 by bsongis