opentx/opentx #8274: Horus X 12S does not work with Multimodul Intern + Extern in Version 2.3.11

Githubbachelor, 2021-02-05:
Hallo I have many Planes and other Models where I used years over a Multi Modul Jumper and the internel XJT FRSKY System from a Horus X 12S. Today I want to fly my ICON A5 with 13 Channel and I got bind my D8R II Plus with internal XjT D8 Mode FCC Non EU Open TX 2.3.11 an Multi Modul latest Versin Software. I want to bind my second Receiver a DSM2 but I cannot bind the Receiver with the Multi Modul Jumper. Than I tray another RX FLYSKY and also problem. No Receiver I can bind as second Receiver. But everything has worked before?!
If I use my external XJT Modul I can bind a second D8R II PLUs. And If I use my Radiomaster with second Multi Modul I can work with different Receiver. But with my Horus it does not longer Work.
Dear Pascal, can you please make the Multi Modul work with Horus 12S internal XJT D8R II Plus and extrenal Multi Modul Jumper again, so that I can bind the Receiver in that Modul as a second and give them Channel 9-17 in the second Multi Modul and 1-8 into the internal XJT Horus X 12S?
I have 8 Models which has worked before and now nothing??? What happen. Please help to reset the Multiprotokoll that it it work again?
Greetings Michael
Look pictures ...
[image]
[image]

kilrah, 2021-02-05:
Disable the internal module, while you bind the DSM RX. Then reenable it, and disable telemetry on the multi.

closed 2021-02-05 by kilrah

Githubbachelor, 2021-02-05:
Hallo Andre, I tested this and that work ONE TIME. So if you want to need your model you have to bind every time the Multi modul DSM Receiver and later the D8R Receiver. And you have every time to do this if you want to fly with using 2 different receiver. And every time Binding procedure? The problem is that if you switch off your Horus 12S and take away BEC - Switch on TX and BEC to Receivers both that only the D8r will bind again. And you have to do the whole procedure again. That cannot a good result - If I try my Radiomaster with two different RX it will work. But my best System at all is ma Horus X 12S.

What is the problem that the Multi Modul will not give Signal after Reboot to the 2. RX?

kilrah, 2021-02-06:
This is not related to OpenTX.

Githubbachelor, 2021-02-06:
Ok Andre so the Problem is to find in The Software Multimodul? Can you maybe fix that or ask Pascal to find a way for an Multimodul update. I have 3 friends and we have together 23 models where we use this way with two Receiver.  Until 6-10 month ago our System work perfectly and we wonder now why we could bint and lost after every binding and the Multimodul gives no signal out After Horus and On again.