opentx/opentx #8276: AGAIN Open TX 2.3.11 with Multi M. v1.3.2.12 Horus X 12S does not work with Multimodul Intern + Extern After reboot lost signal to Extern - All Receivers

Githubbachelor, 2021-02-06:
Pascal means that it is a open TX problem and that is also what I think! He means if it work in Radiomaster, than it can only be a problem in Open TX. You can read his answer at end here.

Hallo I have many Planes and other Models where I used years over a Multi Modul Jumper and the internel XJT FRSKY System from a Horus X 12S. Today I want to fly my ICON A5 with 13 Channel and I got bind my D8R II Plus with internal XjT D8 Mode FCC Non EU Open TX 2.3.11 an Multi Modul latest Versin Software. I want to bind my second Receiver a DSM2 but I cannot bind the Receiver with the Multi Modul Jumper. Than I tray another RX FLYSKY and also problem. No Receiver I can bind as second Receiver. But everything has worked before?!
If I use my external XJT Modul I can bind a second D8R II PLUs. And If I use my Radiomaster with second Multi Modul I can work with different Receiver. But with my Horus it does not longer Work.
Dear Pascal, can you please make the Multi Modul work with Horus 12S internal XJT D8R II Plus and extrenal Multi Modul Jumper again, so that I can bind the Receiver in that Modul as a second and give them Channel 9-17 in the second Multi Modul and 1-8 into the internal XJT Horus X 12S?
I have 8 Models which has worked before and now nothing??? What happen. Please help to reset the Multiprotokoll that it it work again?
Greetings Michael
Look pictures ...

All actual software and firmware is installed. Yes the extern LED is red flashing and I can only bind the extern Modul of the intern is off. I can bind the intern first than reboot the Horus. After switch intern to off. Than Choose extern and take Multi DSM 21F for Channel 9-14. After I can bind and try telemetry off. Than when is binded I switch intern on again to my RX D8R and it work with FRSKY. Than both are working. But If you switch off HORUS and on again the extern Modul does giv nothing out (NO LED RED) and the extern RX does not work only intern. If want to work I must do that procedure completely again but it can not be that that is a good result. If I want to use I need hours to open the model for taking receivers. Same is with your latest Version 1,3.2.30

@pascallanger
 
Owner
pascallanger commented 11 minutes ago • 
If there is no red led it means simply that the issue is in OpenTX while powering up the module end of it.
I bet the issue has started after you have upgraded OpenTX.

@Githubbachelor
 
Author
Githubbachelor commented 10 minutes ago
The RED LED is working only when the extern Module is working alone or as I explain my procedure

[image]
[image]

pascallanger, 2021-02-07:
Can you do a video of the module when the radio is starting?

Githubbachelor, 2021-02-07:
ok thank you. I will make a video for him.

pascallanger, 2021-02-07:
No need for a video anymore. Debugging is in progress.

Githubbachelor, 2021-02-07:
yes but Andre asked me now.

kilrah, 2021-02-07:
I didn't.

Githubbachelor, 2021-02-07:
Oh sorry it was Pascall email. But I did a video now. So no problem I will give it to you and pascall.

Githubbachelor, 2021-02-07:
Did you see it? Than I delete it after.

closed 2022-04-17 by stale[bot]