opentx/opentx #8278: Relative positioning mode for Joystick/thumbstick Camera- possibly with gradient/expo motion ramping

salukikev, 2021-02-09:
I added a thumbstick to my Taranis X9D+ and rerouted two of the potentiometers to it.   I removed the spring return to center so that I now have absolute positining of a (servo) gimbal which I use to steer a camera.  In this way the joystick mirrors the position of the camera and I can leave it or move it as desired and it will stay put.  This works- but delivers a fairly undesirable herky-jerky motion of the camera when moved.  That is only exacerbated by fast digitial servos.

A vastly improved solution would be to have relative positioning of these camera servos.  This means that I could leave the spring return to center mechanically in place and the camera could move in a much more controlled and gradiated fashion- with the distance from center determing the rate.  These servos would continue to move to their specified limits.

Even if only relative positining were implemented it would be a great improvement- ramped acceleration would be absolutely great.   A feature like this would be very useful on OEM Transmitters where one stick can control the camera and the other ground activity.  You could also use this effectively with knob or lever auxiliary controls- not just a hardware modified unit like mine.   There are many scenarios where a custom servo-based gimbal is the best solution due to size, geometry or other custom constraints- not just the standard OEM multicopter/stabilization gimbal.

Thanks for any consideration!

hrenz, 2021-02-10:
the solution is an integral-mixer
look at german OTX manual there are some examples with integral-mixers with relativ speed
for Translation use DeepL

salukikev, 2021-02-10:
I'm eager to take your suggestion!  thanks!  I guess I'd adapt the request here to maybe make to more obvious/user friendly?   Seems like a bit of a roundabout way for me to try to glean how this is accomplished, but I very much appreciate your help!

salukikev, 2021-02-11:
I'm not super familiar with github yet, so maybe the link is obvious to those who are.   That said, you may have uploaded this as a jpg for compatibility reasons but it's the hardest possible format for me to actually deal with- I even tried running that jpg through an OCR so that I could then try to translate it.   See how that suggestion about making it more user friendly comes into play?   I'm sure I'll get it eventually and thanks again for posting this info.

hrenz, 2021-02-12:
Hy,
no problem, 
here its aword-docx
or load the complete manual at frsky-forum  at Engel Modellbau German, its from me
 https://frsky-forum.de/lexicon/index.php?entry/138-handbuch-f%C3%BCr-opentx/about 130MB 1006 pages, Mai 2020

mfgH.Renz

Verschickt: Fr, 12. Feb. 2021 0:34

salukikev, 2021-02-12:
Here is my effort at translation- some of the graphics didn't register correctly so I tried to replace the relevant text in photoshop and then overlay.  I think this is right, and it's only the two pages being discussed.  [Here is a link to PDF](https://drive.google.com/file/d/1xeCPEKt74D0XlbxSC10ejU_0lJJe7q3R/view?usp=sharing).  [Here is a link to Document](https://drive.google.com/file/d/19WGHL7hJFr_eLejZZaiP7dw02zFrUuIF/view?usp=sharing).

marcusluciano, 2021-02-25:
I was over on Reddit talking about nudging global variable via a script.  I went to plug it into the only opentx radio I have left in stock, accidentally put the battery in backwards, and fried my Taranis.  I have no idea when my parts will arrive :(

In the meanwhile, here is the untested script.  The .lua file goes into ./SCRIPTS/{modelname} folder, the nudge.txt into the root (I think).

[nudge.zip](https://github.com/opentx/opentx/files/6046003/nudge.zip)