opentx/opentx #8288: Telemetry 4+2 without sliders causes radio to crash

bitogre, 2021-02-14:
**Describe the bug**
Get "Emergency Mode" screen when trying to setup widgets in 4+2 with sliders disabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Go into Telemetry on Radiomaster TX16S
2. Select 4+2 widgets
3. Disable Sliders
4. Setup Widgets
5. "Emergency Mode"

**Expected behavior**
Able to setup the various widgets

**Video**
https://photos.app.goo.gl/ELboEAUj9pvbUABG9

**General information**
 - OpenTX version 2.3.11
 - Hardware: Radiomaster TX16S Color

3djc, 2021-04-25:
Cannot duplicate, seems ok here doing the same

bitogre, 2021-04-25:
Give me until May 9th to get you my model file (I am about to go out of town to attend https://seffweek.com/) in case it is specific to the model that is loaded when it happens.

bitogre, 2021-04-25:
Here is my backup file and otx file that I am able to reproduce it 100% of the time with the "CX5" model in the DLG folder.  It happens when changing Telemetry page 1.
[Devin.zip](https://github.com/opentx/opentx/files/6373456/Devin.zip)

I also wonder if the theme makes a difference.

3djc, 2021-04-26:
Doesn't happen hjere. I'm puzzled why you model bin file is 32k tho, should be more like 10k

kilrah, 2021-04-26:
The 32k file is an X9D+ eeprom...

bitogre, 2021-05-02:
> The 32k file is an X9D+ eeprom, not sure how it's related...

All the models on my Radiomaster TX16 were ported over from my Taranis X9D+.  Looks like I mixed up my last backup on the Taranis with the first backup with the Radiomaster.

bitogre, 2021-05-02:
> Doesn't happen here. I'm puzzled why your model bin file is 32k tho, should be more like 10k

Did you try using the .otx file?  Or do I need to create a new Radiomaster backup file?  Or is there something else I can provide that might help with reproducing?

bitogre, 2021-08-21:
Looks like others have seen this issue and reported in #8642

3djc, 2021-08-25:
Could you try tomorows nightly (557 or later) ? I just pushed a fix that 'might' solve this

bitogre, 2021-08-29:
OK, I cannot reproduce on 2.3.15N560.  BTW, I have had harder times reproducing it lately.  I suspect the issue is related to how I moved my models from my Taranis X9D to the RadioMaster.  It seems to only show up on models I have not used since moving to the RadioMaster.  I am not able to reproduce on 2.3.11 on models that I have used a lot and updated multiple times since the transfer.

3djc, 2021-08-30:
I think I found an issue when you had extented trims and no sliders

daveelsey, 2022-11-02:
I have a new X10S that has done this a few times, seems to be when playing with widgets. Not bricked, just shut down, then repower and its back to normal. I hope this can't happen when flying as there is no control when in this emergency mode.

3djc, 2022-11-02:
You should have control in emergency mode