opentx/opentx #8289: Companion compare of model's checklist settings wrong for TX16S

drtrigon, 2021-02-14:
When using companion with TX16S and comparing the model's checklist, 2 issues arise:
1. if the checklist displaying option is off the checklists are not displayed in compare window - but on the TX they can be displayed and are actually existing, it's just they are not displayed after switching model or the TX on
2. comparing the checklists does not work, as both are loaded from local SD storage which of course leads to always showing no differences

**To Reproduce**
Steps to reproduce the behavior:
1. Use companion set up for TX16S
2. Use a model with checklist but disable the option to display it (when switching model or TX on)
3. Use the compare models function to look at the model
4. See issues

**Expected behavior**
1. whenever a checklist exists the comparison window must display it
2. use consistent state of checklist with acording model settings - this might be very hard I understand that - but then I is even more important to solve the first issue in order to be able to "compare" it manually by using your memory

**Screenshots**
-

**General information**
 - OpenTX version: 2.3.9 (the one it came installed with)
 - Hardware TX16S
 - OpenTX Companion: 2.3.11

3djc, 2021-02-14:
I'm not sure I understand. Settings for inactive items are never shown, and should  not be IMHO

drtrigon, 2021-02-14:
Yes, my point is that creating a checklist makes it active already. I can long-press the roller button and choose to show notes and get the checklist displayed. The checkbox just defines whether the checklist is displayed (without me asking for) when selecting a new model or powering the TX on.
So these are actually 2 settings:
1. have a checklist to confer when I open a menu and select to show it to me
2. an option that makes the checklist show up when I change model or power on

3djc, 2021-02-14:
The checklist display is merely a shortcuts to display a text file on your SD, that you could do as well with sd explorer, it does not 'do' anything

drtrigon, 2021-02-14:
I do not get this. The checklist is set-up and can be used so it should also be displayed and compared. This is not related to the option "show the checklist w/o the user asking for".

I consider it as a feature (not a bug) having a checklist defined and available that DOES NOT get displayed if I do not ask for it. Still I need it occasionally. Consider for example a checklist that contains some hints on how to bind a model. This is something I do not need to use every time I switch the TX on but only after I did not use the model for long time and come back and forgot how this strange procedure was... So it is annoying to get the checklist displayed every time, but it is crucial to have it in case I need to consult it.

drtrigon, 2022-01-05:
1. I am still of the oppinion that there should be a shortcut display on otx whenever it present, using th sd explorer is cumbersome as it are way more clicks (depending on the name of the model). I would like to ask whether its possible to add this as a new feature.

2. I also still think the comparison for changes of the checklist is broken or not existent in companion but would be very useful if available and thus I would like to ask for this as a new feature.