opentx/opentx #8290: Cannot select telemetry sensor for warning on TX16S

drtrigon, 2021-02-14:
**Describe the bug**
I did a discover sensors on the TX and got a list of sensors. After that I would like to set the sensor on the telemetry page as RSSI source. That is not possible, I can select the source field but when I move the roller button I just get beeps and no entry other than "(default)" shows up. Now the intressting thing is I can change the value in companion.

**To Reproduce**
Steps to reproduce the behavior:
1. Discover telemetry sensors
2. Try to set one of these sensors under source (RSSI) in order to enable alarm for that value
3. See error

**Expected behavior**
A field/list should appear (like e.g. for switch selection on mixer page) that contains the discovered sensors in order to select one. This is what happens in companion.

**Screenshots**
-

**General information**
 - OpenTX version: 2.3.9 (the one it came installed with)
 - Hardware: TX16S
 - OpenTX Companion: 2.3.11

3djc, 2021-02-14:
Working as intended. If you want custon alarm, do them using special function and logical switches

closed 2021-02-14 by 3djc

drtrigon, 2021-02-14:
Ok, so this is a bug in companion. ;) I think we should keep it open in order for the companion bug to be solved? I can change the title and description if you like?

That also means I missunderstood the function of this "source" field. What is it for then?

3djc, 2021-02-14:
A bug as been open for this : https://github.com/opentx/opentx/issues/8291

RSSI is to select the RSSI source when you have several modules. (basically internal or external)

drtrigon, 2021-02-14:
Ok, perfect! Thanks for the info!