opentx/opentx #8294: Building firmware on Windows fails for Horus radios at generating datacopy.cpp for ./gui/480x272/layout.h:23:10: fatal error: 'list' file not found

rotorman, 2021-02-14:
Followed Windows Build setup guide [here](https://github.com/opentx/opentx/wiki/OpenTX-2.3-Windows-Build-Instructions).
Builds OK _companion_ and _simulator_, but not _firmware_ for Horus-class radios. GCC for ARM installed and it's bin folder set in PATH as first folder.

Output while building firmware:
```
...
FAILED: radio/src/datacopy.cpp 
cmd.exe /C "cd /D C:\Users\rotorman\Desktop\OfficialOpenTX\radio\src && "C:\Program Files\Python39\python.exe" C:/Users/rotorman/Desktop/OfficialOpenTX/radio/util/generate_datacopy.py datastructs.h -DPCBHORUS -DPCBX10 -DCPUARM -DSTM32F4 -DSTM32F429_439xx -DCOLORLCD -DBACKUP -DSIMU -I. -Itargets/horus -Itargets/common/arm/stm32 -Ithirdparty -Ithirdparty/STM32F4xx_DSP_StdPeriph_Lib_V1.4.0/Libraries/CMSIS/Device/ST/STM32F4xx/Include "-Ic:\program files (x86)\gnu tools arm embedded\4.7 2013q3\bin\../lib/gcc/arm-none-eabi/4.7.4/include" "-Ic:\program files (x86)\gnu tools arm embedded\4.7 2013q3\bin\../lib/gcc/arm-none-eabi/4.7.4/include-fixed" "-Ic:\program files (x86)\gnu tools arm embedded\4.7 2013q3\bin\../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/include" > C:/Users/rotorman/Desktop/build-OfficialOpenTX-Desktop_Qt_5_15_2_MinGW_32_bit-Release/radio/src/datacopy.cpp"
./gui/480x272/layout.h:23:10: fatal error: 'list' file not found
```

Failed line of layout.h:
https://github.com/opentx/opentx/blob/e94f6c0ccdc1ecd800f2ef27a28120878bc9ec41/radio/src/gui/480x272/layout.h#L23

I cannot further circle down the error, as why class list cannot be found, as under following folder I have list.tcc template:
_C:\Program Files (x86)\GNU Tools ARM Embedded\4.7 2013q3\arm-none-eabi\include\c++\4.7.4\bits\list.tcc_

As the code builds just fine under Ubuntu 20.04, I assume this not to be an issue in C++ nor header files. Taking from RCGroups discussion, I am not the only one with the problem. Could someone please look into it? Thank you!

Mutantodon, 2021-02-22:
I just had the same issue trying to compile the OpenTX 2.3 branch.
I resolved the problem (and more compile errors) by adding two system environment variables for the GCC include paths:
`C_INCLUDE_PATH=C:\Program Files (x86)\GNU Tools ARM Embedded\4.7 2013q3\arm-none-eabi\include` and
`CPLUS_INCLUDE_PATH=C:\Program Files (x86)\GNU Tools ARM Embedded\4.7 2013q3\arm-none-eabi\include\c++\4.7.4;C:\Program Files (x86)\GNU Tools ARM Embedded\4.7 2013q3\arm-none-eabi\include\c++\4.7.4\arm-none-eabi`
After this the firmware compiled successfully.

rotorman, 2021-02-24:
Thank you for the tip, greatly appreciated, I will look it up (soon). I am currently doing a [in-circuit-debug guide under Ubuntu](https://www.rcgroups.com/forums/showthread.php?3836727-Blog-2-Exploring-and-debugging-OpenTX-with-Qt-Creator-IDE) for OpenTX fw running on radio µC, but if I get the Windows version running, I will re-visit this under Windows.

rotorman, 2021-02-27:
@Mutantodon I followed the instructions [here](https://github.com/opentx/opentx/wiki/OpenTX-2.3-Windows-Build-Instructions) almost to the point, with the difference that instead of MSVC 2017 32-bit, I installed MSVC 2019 32-bit as possible to choose in latest LTS Qt 5.15.2) and also tried your suggested added environment variables, this did not bring success, as I still get:
```
File "C:/opentx/van2311/radio/util/generate_datacopy.py", line 128, in <module>
    main()
  File "C:/opentx/van2311/radio/util/generate_datacopy.py", line 116, in main
Scanning dependencies of target x10_alpha_bitmaps    index = clang.cindex.Index.create()
  File "C:\Program Files (x86)\Python38-32\lib\site-packages\clang\cindex.py", line 2699, in create
    return Index(conf.lib.clang_createIndex(excludeDecls, 0))
  File "C:\Program Files (x86)\Python38-32\lib\site-packages\clang\cindex.py", line 212, in __get__
    value = self.wrapped(instance)
  File "C:\Program Files (x86)\Python38-32\lib\site-packages\clang\cindex.py", line 4147, in lib
    lib = self.get_cindex_library()
  File "C:\Program Files (x86)\Python38-32\lib\site-packages\clang\cindex.py", line 4178, in get_cindex_library
    raise LibclangError(msg)
clang.cindex.LibclangError: [WinError 193] %1 ist keine zulässige Win32-Anwendung. To provide a path to libclang use Config.set_library_path() or Config.set_library_file().

```
By looking at line 4147 of cindex.py I see it fails to load _libclang.dll_, which I did copy, as in Step 2 of the guide, to C:\Windows\SysWOW64.

Thus we must be doing something different if you are successful. Would you care to list your differences to the guide [here](https://github.com/opentx/opentx/wiki/OpenTX-2.3-Windows-Build-Instructions)? What version of Python are you using (I tried 3.9.2 x86, 3.8.8 x86, 2.7.12 x86, 3.9.2 x64)? Where did you possibly additionally copy _libclang.dll_? Thanks!

Mutantodon, 2021-02-27:
I installed Qt 5.15.2 with MSVC 2019 32-Bit and MingW 8.1 32-Bit. Python version 3.9.2 x64.
I remember I had to put _libclang.dll_ in _C:\Windows\system32_ folder instead of SysWOW64 subfolder. That should make the error go away.

rotorman, 2021-02-27:
@Mutantodon - A big thank you to you! It works now like a charm.

The [GitHub Windows compilation Wiki page](https://github.com/opentx/opentx/wiki/OpenTX-2.3-Windows-Build-Instructions) could use the following update:

- Section 2 Point 3 -> MSVC 2019 32-bit
- Section 2 Point 4 -> MinGW 8.1.0 32-bit
- Section 2 - copy _libclang.dll_ also on 64-bit systems to _C:\Windows\System32_ and NOT to _SysWOW64_
- Section 4 - on x64 systems, install x64 version of Python and there is no reason not to use the latest 3.9
- Section 4 - the paths for Python are automatically added, if this is selected in _Custom installation_ method, so no need to do it manually
- Section 8 - use following URL instead to get LibSDL v2: [http://www.libsdl.org/download-2.0.php](http://www.libsdl.org/download-2.0.php)
- Need to add two system variables:
```
C_INCLUDE_PATH=C:\Program Files (x86)\GNU Tools ARM Embedded\4.7 2013q3\arm-none-eabi\include
CPLUS_INCLUDE_PATH=C:\Program Files (x86)\GNU Tools ARM Embedded\4.7 2013q3\arm-none-eabi\include\c++\4.7.4;C:\Program Files (x86)\GNU Tools ARM Embedded\4.7 2013q3\arm-none-eabi\include\c++\4.7.4\arm-none-eabi

```
After these changes I was finally able to compile also under Win 10 x64-bit without problems.

I will close this issue, after I manage to get the Wiki updated.

rotorman, 2021-02-27:
The major issues are now fixed in Wiki thanks to @3djc. Closing!

closed 2021-02-27 by rotorman