opentx/opentx #8299: Receiver list

MrD-RC, 2021-02-15:
It would be a really useful feature to be able to see which receiver numbers have an attached (bound or registered) receiver associated with it. 

Currently, we would have to keep notes off of the transmitter. Or just register/bind and hope there are no conflicts. A page showing the status of the receiver numbers would be really handy. ACCST receivers could just show *Empty* or *Allocated*. Whereas ACCESS the name of the receiver could be shown. 

Other useful information on this screen could be the models that the receivers are attached to. 

I imagine this *receivers* screen would be added to the system menu.

3djc, 2021-02-16:
It is done on companion, but cannot quite be done on the radio. But the radio will tell you is an rxnumber is already in use. Also rxnum and access receiver name have nothing in common. You can all all your receiver "RECV" if you so which

MrD-RC, 2021-02-16:
Do you not register your ACCESS receiver to an rxnum? In which case, the name could be used to let you know which receiver is attached. Is that registration is then transmitter wide? It’s only the bind that associates the receiver to the model?

I understand that all receivers could be called the same name. But I also see how that would be a waste of functionality. The only issues I could see is if the names aren’t stored on the transmitter.

I don’t really understand why Companion can  show the data, but the transmitter couldn’t. In companion, the only thing I’ve seen is the txnum next to the model name in the list when you load models and settings from the radio. This leads me to believe that all the data to make this list is already on the transmitter? So I don’t quite understand why a system **receivers** screen couldn’t be implemented in the future?

kilrah, 2021-02-16:
> I don’t really understand why Companion can show the data, but the transmitter couldn’t. 

Companion has all models loaded at the same time, the radio only has one and doesn't know anything about the others apart from the rxnum that is stored in a global list.

MrD-RC, 2021-02-19:
So why couldn’t the receiver being in use or not also be in a global list? I understand that it’s not curently.

Also, how does this work with ACCESS? I was under the impression that registration (which is associated with the rxnum) is transmitter wide. It’s only the bind that is on a per model basis.

kilrah, 2021-02-19:
Excessive memory usage.

ACCESS registration is unrelated to the radio, it's about the receiver storing the ID of the radio it's allowed to bind to.

pfeerick, 2021-03-15:
Would this be such a problem if it were a system menu, and the model names attached to 'in use' receiver IDs loaded on demand? 

i.e. A list of all the in use receiver IDs, with a option to load the associated names from the model memories?

I'm glad I'm not the only one finding the receiver number cumbersome. Even something as 'simple' as auto-increment to the next available slot would make life easier... as it gets pretty ordinary figuring out what RxID number to use when adding another model on the Tx. Meaning the Companion gets a lot of use (although it would not be fair to say for that reason alone! :laughing: )

closed 2022-04-17 by stale[bot]