opentx/opentx #8300: Center beep not consistent

MLeo84, 2021-02-16:
Hello, I've activated center beep for my throttle stick (in an attempt to get a better feeling at which position I'm actually at center). 
If moving the stick quickly back and forth, not everytime the stick crosses center, an actual beep gets played. Further, there seems to be a delay between actually crossing center and playback of the beep. And even worse, this delay doesn't seem to be consistent.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate "Center Beep" for throttle in "Model Setup" menu
2. Set "Play delay (sw. mid pos)" in "Radio Setup" to 0ms
3. Move throttle stick up/down quickly around center
4. Observe inconsistent beeps

**Expected behavior**
A clear beep at every crossing of stick center with minimal (but definitely consistent) delay.

**General information**
 - OpenTX version 2.3.9-otx (60e1edc1)
 - Hardware Radiomaster TX16S
 - OpenTX Companion not used, yet

closed 2022-04-17 by stale[bot]

nickthenorse, 2025-02-10:
This is still an issue in v2.10.6

gagarinlg, 2025-02-10:
There is no OpenTX 2.10.6

nickthenorse, 2025-02-10:
> There is no OpenTX 2.10.6

Hehe very true!

I read - and commented - too quickly. I meant this is an issue on EdgeTX v2.10.6.

I'll see myself out.