opentx/opentx #8302: read and import losing channel output max

guillaumevaret, 2021-02-16:
only on companion 2.3.11, windows
I don't have the problem on previous version (2.3.9) 

When I read a model from the radio X10S express or from  a file, with a channel output max VG, its losing the VG info, converting it in 150%. The VG info on min is not lost.

elecpower, 2021-02-18:
On Windows 10 and Ubuntu 20 tested with Companion 2.3.11 a new X10S Express profile and new model. Changed sub-trim, min and max to different GVs. Closed and saved model file, reopened fie and model, displayed GVs as saved.

3djc, 2021-02-19:
@guillaumevaret maybe you could provide said file ?

guillaumevaret, 2021-02-20:
yes. And with a screen view of the difference before saving the file / after opening the file
[image]

[pb vg 11.zip](https://github.com/opentx/opentx/files/6014598/pb.vg.11.zip)

elecpower, 2021-02-20:
Bug confirmed. Triggered by extended limits being selected in model setup

PR #8321 (2021-03-10): Stop extended limits affecting output fields referencing global variables in Companion
Fixes  #8302
closed 2021-03-10 by bsongis