opentx/opentx #8303: Add notes field to each model

MartyHugh, 2021-02-16:
**Is your feature request related to a problem? Please describe.**
Some Logical Switches logic is multiple lines chained together.  Often lines refer to other logical conditions which are not immediately close by and have to be scrolled to to examine.

**Describe the solution you'd like**
What would be really useful would be to have a "notes" field in each  model which helps inform a user so they can go back and debug or modify an existing logical switch.  Often these things are much easier to write than they are to read, especially if it is many months later on.

**Describe alternatives you've considered**
Add a notes field to each model.  This way it need not be specific to logical switches, it can also be used to explain some arcane mixes or other features, assumptions or limitations of a model.

kilrah, 2021-02-16:
The checklist feature could probably be used for that

MartyHugh, 2021-02-16:
Thanks kilrah. Is this feature available for editing in Companion ?

MartyHugh, 2021-02-16:
I have found this in Companion.

MartyHugh, 2021-02-17:
I think the checklist feature is a pretty close fit.  The downside of it compared to what I had in mind is that the notes fields are held externally to the models and settings. Ideally this field would follow a model around just like all the other settings held in EEPROM.  Obviously the trade off would be EEPROM space and the number of models one could store.  

I have around 15 models in EEPROM which produce an exported size of around 750 bytes per model.  Assuming a maximum of 32Kb (I understand this is what the X9D has, do other radio models differ?),  and allowing up to (say) 240 bytes for a notes field, this would still allow for more than 30 models to be held in EEPROM.

This would be a compromise I would be very happy to accept given that a) I dont like the idea of having to maintain external files to an exported model and settings file, and b) I cant see my own case getting anywhere near 30 models for immediate use in EEPROM at any one time.

Thank you for reading.

3djc, 2021-02-17:
We have plenty of people with EEprom space issue already, so this simply cannot be added to EEprom, especially that 240 would be too little anyway to actually comment on the rational for LS or complex programming anyway

MartyHugh, 2021-02-17:
Thank you for your comments 3djc,  I agree that 240 bytes might require some journo-short hand skills in places.  If you are saying that there is a significant number of users who are already having issues with EEPROM space then this is clearly not a viable option.

MartyHugh, 2021-02-17:
Issue closed due to hardware constraints.

closed 2021-02-17 by MartyHugh

lshems, 2021-02-21:
Marty,
Entering text is not a nice thing on a radio without keyboard. But creating a text file that loads on startup on a pc and put it in place is easily done.

Copying that file CAN be done on the radio sdcard browser, if I'm correct.

If you set up your text file to be able to change just the switch assignments for instance, that would become feasable on the radio. Changing SC to SD in a text file is doable.

There exist solutions for editing the checklist text file on the radio, and show it from the telem screen from instance using Lua scripts.

MartyHugh, 2021-02-21:
Very interesting Ishems.  I would be grateful for any links to further information on this.

lshems, 2021-02-21:
Check my blog on rcgroups, or pm on rcgroups.

L Shems