opentx/opentx #8304: Mac: OpenTX Companion wants permission to listen for keystrokes from ALL applications

DavidJJ, 2021-02-16:
**Describe the bug**
MacOS Catalina and Big Sur

On first launch, OpenTX Companion requests permission to listen for keystrokes from all and any applications. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to OpenTX Companion Downloads
2. Click on Download latest version
3. Open application.
4. See permissions request dialog.

**Expected behavior**
I see no reason why the OpenTX companion software should need or want to listen to keystrokes meant for other applications.  Even if innocuous, this is something a keylogger would do.    

Denying the request does not seem to affect the application so the permission is not needed. 

**Screenshots**
[image]

**General information**
 - OpenTX version (N440)
 - Hardware (e.g.Jumper T-Lite)
 - OpenTX Companion (e.g. 2.3.12 on MacOS)

3djc, 2021-02-16:
Clearly not a macOS expert by I seem to recall it is required for some of the simulator inputs

kilrah, 2021-02-16:
It's required by the simulator for it to respond to keypresses without any text entry field and whether its window is in focus or not.