opentx/opentx #8307: stackAvailable() returns always 0

rotorman, 2021-02-17:
By testing OpenTX with DEBUG option built in on TX16S, I see that physically running the OpenTX code on STM32F429BI, _stackAvailable()_ returns always 0 for the main task.

Looking at the implementation of stackAvailable():
https://github.com/opentx/opentx/blob/eaa972cd2fbe2db5b975661227ba92022a752449/radio/src/rtos.h#L226-L229

https://github.com/opentx/opentx/blob/eaa972cd2fbe2db5b975661227ba92022a752449/radio/src/rtos.h#L209-L217

I can see that it looks for the _paint_ pattern 0x55555555, but as it returns 0 is unable to find from the end of stack (=lowest address).

On the other by looking at the Reset_handler I see that it should be filled correctly:
https://github.com/opentx/opentx/blob/eaa972cd2fbe2db5b975661227ba92022a752449/radio/src/targets/horus/startup_stm32f42_43xxx.s#L104-L112

Is really main task stack used up to last byte, or have I misunderstood it or is there (just maybe) an implementation bug?

rotorman, 2021-03-01:
I had a look with in-circuit-debugger into the memory and indeed no _paint_-pattern 0x55555555 can be found at _main_stack_start 
[image]
address:

rotorman, 2021-03-01:
And the _main_stack_stark is nicely "painted" on reset as can be seen here:
[image]

rotorman, 2021-03-05:
For testing purposes, I changed in linker script https://github.com/opentx/opentx/blob/f8498ddf86f37ecf4a1eb962dec0776d220984dc/radio/src/targets/horus/stm32f4_flash.ld#L28 main stack size to 2048 bytes and what I can see, is indeed, that the main task stack usage on TX16S is slightly over default 1024 bytes (values less than 0x1000FC00 are not painted anymore with 0x55555555, where the stack top is at 0x1000FFFF and 1K default border for STM32F4xx Horus class radios would be 0x1000FC00):

[image]

@3djc , @kilrah would you care to look into this?

rotorman, 2021-03-28:
@bsongis  Were you able to replicate the issue? Thanks!

raphaelcoeffic, 2021-04-06:
@rotorman when you crash the stack, you really notice it, believe me ;-) It's most probably a bug in that function.

rotorman, 2021-04-06:
I increased the stack to 2048 (see above), I see that it is used over the first 1024 byte border, thus the most important message to me here is that 1K, as by default, is not enough.

As I know it - if the memory usage (starting from low addresses towards high addresses in memory; stack is opposite, from high addresses towards low addresses) never gets used up to to stack end (lowest stack address), then you won't notice it, as there is no clash.

Out of curiosity - what function are you referring to? Initial stack paining in assembly and _getStackAvailable()_ that searches for the paint pattern starting from lowest address, both look fine to me.

closed 2022-04-16 by stale[bot]

bsongis, 2022-12-16:
This issue should not be closed. 
It doesn't crash because the area just before the stack is not used by anything!