opentx/opentx #8310: Wrong telemetry warning values in companion

drtrigon, 2021-02-18:
**Describe the bug**
The telemetry low and critical values of a model on the TX are not handeled correctly by companion. When reading the models with companion and going to the telemetry page I see just the default values of 45 and 42.

**To Reproduce**
Steps to reproduce the behavior:
1. On the TX on the telemetry page change the low and/or critical values to something other than (45 or 42)
2. Connect the TX to a PC and use companion to read the setting
3. Open the models telemetry page on companion
4. See error

**Expected behavior**
Companion should use the values given in the model (by the TX).

**Screenshots**
(n/a)

**General information**
- OpenTX version: 2.3.9 (the one it came installed with)
- Hardware TX16S
- OpenTX Companion: 2.3.11

elecpower, 2021-02-18:
Tested on Windows 10 with my TX16S with 2.3.11 firmware and Companion. Companion shows the RX adjusted values. Also tested changing values in Companion, saving model and re-opening and changed values shown.

3djc, 2021-02-19:
Cannot replicate either.

drtrigon, 2021-02-19:
I use companion in linux, Kubuntu 20.04.

Tested again, I set critical to low to 20 and critical to 15. When I look at the telemetry page I see 20 in both fields. So I have to correct the bug report, it does show one of the two values...

elecpower, 2021-02-19:
Confirmed Companion ranges do not match radio.

drtrigon, 2021-02-20:
Thanks a lot!

PR #8315 (2021-03-10): Mirror rssi alarms in Companion
Fixes #8310
closed 2021-03-10 by bsongis