opentx/opentx #8311: Throttle stick position alarm voice audio output strange in German

drtrigon, 2021-02-18:
**Describe the bug**
Audio output of the throttle stick position is somehow messed up in German as it is barely understandable (at best). Actually it took me several days until I recognized that this strange voice output is meant to be a trhottle stick position warning. I cannot even describe it properly.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure to use German voice audio output
2. Enable throttle stick position warning
3. Set throttle stick to something else than -100%
4. Turn the TX off/on or change model in order to trigger the warning
5. Hear error

**Expected behavior**
I think the message is meant to be something like "Gas nicht null - bitte prüfen!".

**Screenshots**
(n/a)

**General information**
- OpenTX version: 2.3.9 (the one it came installed with)
- Hardware TX16S
- OpenTX Companion: 2.3.11

drtrigon, 2021-03-09:
Also the startup voice is wrong as it sais "Willkommen in Open-T-I-X" instead of "Willkommen in Open-T-X" (without the "I" between "T" and "X").

drtrigon, 2021-03-16:
Assuming there are not many (if any) German-speaking devs here, how can I help?

Can you give me a hint in which source file the relevant code contains? I am willing to create a pull request, but need help navigating and modifying the sources.

jasonleekennedy, 2021-03-16:
Did you check the german voice pack's audio for the 2 times are correct? 

here appears to be the major work behind the audio logic, https://github.com/opentx/opentx/blob/2.3/radio/src/audio.cpp, however that appears to be just playing audio clips off of the SD card.

based on this line:
https://github.com/opentx/opentx/blob/2.3/radio/src/audio.cpp#L179

I suspect you should check the file named thralert and see if it sounds correctly, if not, that is the problem.
and I would guess (but haven't verified) that the audio file named hello is going to have the 'hello' start up audio.

kilrah, 2021-03-16:
https://github.com/opentx/opentx/blob/9693d87c9fb6fd1191dacc16a8202fbd31348d37/radio/util/tts_de.py#L61

drtrigon, 2021-03-16:
Thanks for all the info! I used https://downloads.open-tx.org/2.3/release/sdcard/opentx-tx16s/sdcard-480x272-2.3V0035.zip and tested:
* `SOUNDS\de\SYSTEM\hello.wav`: sounds correct
* `SOUNDS\de\SYSTEM\thralert.wav`: this sound wrong to me, even the rythm so number of syllables is wrong

The next step is to check whether the `hello.wav` on my SD is the same as the one in this pack (size, etc.).

For `thralert.wav` is the question how to proceed. I assume that this file would need to be synthezised again using https://www.open-tx.org/2014/03/15/opentx-speaker and the free eSpeak languages according to https://support.microsoft.com/en-us/topic/how-to-download-text-to-speech-languages-for-windows-10-d5a6b612-b3ae-423f-afa5-4f6caf1ec5d3. The questions here are what voice and settings to choose? The ones from eSpeak seem not to be the right ones (quite robotic, lot of artifacts). I was lucky to have a win install including German voice "Hedda" to test (file attached below). But I don't think this is the right voice nor legal to use.

[thralert.zip, Volume 100, 32 kHz, Microsoft Hedda Desktop](https://github.com/opentx/opentx/files/6152337/thralert.zip)

kilrah, 2021-03-16:
The sounds are autogenerated on every release using sox, not manual.

drtrigon, 2021-03-16:
Are you referring to: http://sox.sourceforge.net/ ?

Did not find any hint on that - thanks for the clarification!

So then I have to check what sox does and whether something bad happens there.

Where does sox get invoked? What parameters?
(I found https://github.com/opentx/opentx/blob/2.3/radio/util/maxvolume.cmd only and this is just normalization if I get it right.)

drtrigon, 2021-03-17:
I need the exact command-line expression in order to reproduce the issue. According to https://wiki.ubuntuusers.de/Sprachausgabe/#SVOX-pico2wave I tried:

```
$ sudo apt install libttspico-utils sox
$ pico2wave --lang de-DE --wave /tmp/Test.wav "Gaskanal nicht Null, bitte prüfen" ; play /tmp/Test.wav; rm /tmp/Test.wav
```

but this is not the same as what I found in `sdcard-480x272-2.3V0035.zip` - so I does not make sence to continue until I know the exact command used (parameters, options, settings, etc.).

What I expect is that it makes a difference whether we use "Gaskanal" or "Gas-Kanal" - and that is what I would like to verify.

drtrigon, 2021-03-17:
I found the reason why the `hello.wav` in otx 2.3.9 (used on the TX16S) sounds wrong: https://github.com/opentx/opentx/commit/70d5ca471def812b40bf055d62242c5de08da205#diff-5566674131383483b6e299a5efd39be3ceb54db91b967c6e82d4edbd1d60ca6c

Still I need help in navigating the code in order to find the place where sox gets called and see how it is called.

drtrigon, 2021-04-28:
@3djc: You made the change 70d5ca4#diff-5566674131383483b6e299a5efd39be3ceb54db91b967c6e82d4edbd1d60ca6c can you explain to me how to test such changes? I was not able to figure out where and how these sound files (or the SD card content) does get build? Can you point me to some ressource? In particular I am intressted in the command line to generate the voice sound files? (I guess it's pico2wave in linux?)

drtrigon, 2021-05-07:
Without further info I cannot contribute to this issue - meaning somebody else has to resolve this.

drtrigon, 2022-01-05:
Today I checked https://downloads.open-tx.org/2.3/release/sdcard/opentx-tx16s/sdcard-480x272-2.3V0038.zip and this sound way better now - will test on the TX16S soon.

Still ... can somebody explain to me where/how this sd card package/zip is built? How can I test such changes?

closed 2022-01-05 by drtrigon