opentx/opentx #8317: Lua information if LCD is on or off

Storchy1, 2021-02-20:
It would make sense to stop the execution of all lcd.draw ... functions if the LCD is off.
This would be very easy if you had a lua function that tells you the status of g_eeGeneral.lightAutoOff.
for example getLCDStatus() I cannot judge how laborious it is to implement such a function.
In any case, it would save a lot of computing time, where no display is on you don't need to perform any draw functions.
Many Thanks

lshems, 2021-02-20:
This is the entire purpose of the background and the run functions ??!

lshems, 2021-02-20:
Or are you talking color screen?
Even then, if it runs with the screen on, it will run without as well. Why saving computing time?

Scripts should work in all conditions.

Storchy1, 2021-02-20:
I am talking from color LCD screens like Horus X12s. I once learned as a software developer that I don't use any unnecessary computing power

lshems, 2021-02-20:
I once learned as a human being to not pay attention to things that are not useful.

No, really, no kidding, what would be the real use for that? If code runs with LCD on, it will run with LCD off. So why bother?

Is there a special reason for this? Just trying to understand.

Storchy1, 2021-02-21:
Hi, 

I will try to describe it to you briefly, as you helped me a lot to understand the LUA script language. 

I do not do this via GITHub because I do not deny that it is allowed, I don't want to annoy anyone. 

I have an X12s. This is now equipped with an external GPS. This GPS gets a fix after max. 30s if the following happens. 

I start X12s then press "MDL" or "SYS" wait until I see the red LED on the GPS and press RTN, then I have a fix.

If I start normally and my widgets run, in which lcd.drawText is called very often, I never get a GPS fix.

I need the GPS because I use it to operate a flight log widget on the Horus. 

I already described the problem with a video in the Horus thread https://www.rcgroups.com/forums/showpost.php?p=45522247 <https://www.rcgroups.com/forums/showpost.php?p=45522247&postcount=17696> &postcount=17696 

So now to the desired function, when I set the timeout for Backlight Mode Duration to 5s and then finish my model Do I get a GPS fix 

I know you don't really need all of this, but I'm currently having fun programming lua. 

Everything is not important except health!

Ciao Jörg

lshems, 2021-02-21:
Ok, that's a very good reason. But also a workaround.

The root cause must be somewhere else.

Why would the GPS not lock if a Lua script is running? Especially the
LCD.draw functions, as they have nothing to do with any GPS interaction.

This points to some other problem, and you already hinted to that: lowering
CPU load.

So now we need to figure out WHAT is not working because of a high CPU load
preventing the lock.

What sensor are you using? Wouldn't it lock without the radio being on at
all? Locking is an on-receiver action, no?

Really strange, and although your proposal could work there are much easier
ways to achieve it perhaps.

Simply put a check on the first line of the run function to check if the
lock is attained. If not, return without doing anything.

Wouldn't that work?

Storchy1, 2021-02-21:
I have a BN-220

I have more the one widget in my Model.

One widget is only drawing a Bitmap, which shows my switches see oxy3SW.png attached

If i start the X12s with this widget as start widget the lock will be in less than 30s.

 

In the moment i have this „first line“ in my run function. It works perfect fort the first lock.

 

But now i get no new coordinates if i call getTxGPS().

So i made for testing a switch to leave run function. --> i get new coordinates, if the switch stands on leave run function.

I write the coords in a file to improve this.

So, how to automate this switch, here comes the new function getLCDStatus().

 

I think i will write a litte function with checks the stick movement, If there is no move with a timeout leave the run function 

I have alway stick movement, i only fly helis.

 

Thanks a lot for your patience.

 

I chose this way to communicate with you, because I don't approve of the choice of words and the hostility in most forums, see e.g. New- FrSky x20 Tandem and ETHOS 

I hope you understand that

 

Ciao

Jörg

lshems, 2021-02-21:
Jörg,

When flying the Heli, you cannot look at the screen, can you?
So I think you can indeed work around it.

Do you still want support for the LCD function, you should clarify the
situation. Otherwise, it doesn't make sense.

Indeed, you ran into the situation where the radio no longer is capable of
supporting the requirement that a script should run no matter what.
I consider it a bug, and your solution a workaround.

I wonder what the Devs will say of it.

Guido

P.S.
I personally also am a little fed up by the agressive tone on much fora.
Actually, the first time I really was upset was due to the opentx Devs.
They are I little more polite lately, but I try to interact as little as
possible.

The ethos thread is a disaster. On the one hand the guys praising frsky for
interacting, on the other hand the guys mad at frsky.

Very difficult to post an opinion there, or an analysis of why what,
without getting shot down personally.

Sad. Indeed.

I do write clear opinions, and do really object to discussions that are
projecting the wrong view that the poor Devs are working for free.

I am perhaps a red tissue to a lot with my opinion.
But it is not decent how some react.

I am constantly trying not to do so myself, and yet I find myself writing a
personal reply every now and than anyway.

Sorry about that.

lshems, 2021-02-21:
Deleted my posts because they ended up here instead of what I thought was a personal mail conversation.

It is not that my unintended posts contained something secret, they where just not intended for public view.

closed 2022-04-17 by stale[bot]