opentx/opentx #8320: X9E > SA is assigned as 3-Pos Switch, but shows the behavior like a 2-Pos Switch

Wesie, 2021-02-21:
OS Vers: 2.3.11
Tanranis X9E

Today I recognize, that the switch SA (It's a true 3-Pos Switch) shows in the Monitor and also in the debug information only the behavior like a 2-Pos Switch. The Switch is assigned as 3-Pos Switch. 
I've open the X9E and changed the HW -Switch agains another 3-Pos Switch, but the result is the same.
I try to program the SA Switch as Toogle and backward to the 3-Pos Switch: Same result.

In the 
- upper position I see the UP arrow >> OK
- middle position I see the UP arrow ( not the (-)) >> NOT OK
- lower position I see the DOWN arrow >> OK

All other installed Switches and Buttons works as expected

3djc, 2021-02-21:
not the case here (I'm a x9e user too). Could you paste your otx file ?

Wesie, 2021-02-21:
[document1.zip](https://github.com/opentx/opentx/files/6018134/document1.zip)

I'll hope, it's the correct one

Wesie, 2021-02-21:
One thing more:
In the HW Menü:
If the SA  switch is up or in the middle, the char of SA are still written in Bold,
If the SA is down (Arrow Down) the letters of SA are no more written in Bold. 
All other Switches are still in every position written in Bold. Has this one a special meaning?

3djc, 2021-02-21:
Works fine, issue is with your radio

closed 2021-02-21 by 3djc