opentx/opentx #8322: How can I change Channel Range

MartyHugh, 2021-02-22:
Once upon a time it was possible to reduce Channel Range from 1-16 to 1-8 to halve the latency.

This does not appear to work the same way now.  Only the starting channel can be selected and changed, and the end channel remains +16 channels away.

What am I missing ?

Thanks for reading,

kilrah, 2021-02-22:
It completely depends on what module/protocol is used so you should say which of the many supported ones you're referring to.

MartyHugh, 2021-02-22:
Yeah good point :-)

I am using FrSky X and FrSky X2, mainly in LBT 8 channel mode.

3djc, 2021-02-22:
So you are only sending 8ch, since thats the way multi works

closed 2021-02-22 by 3djc

MartyHugh, 2021-02-22:
Thanks for the explanation.

3djc, 2021-02-22:
If you want to reduce latency on a multi, you should investigate redpine protocol