opentx/opentx #8329: Bootloader with Radiomaster TX16s and Open tx 2.3.11 (MacBook pro)

Gadget99, 2021-02-27:
Companion does not recognise my radio or SD card. I have been advised that i need to use Bootloader. The only way i can find to start the TX in boot mode is by holding switches 1 & 4 towards the middle whilst pressing the power button. This does not work and the controller boots normally. I have also updated the TX to the latest version of Open TX (2.3.11) via the SD card.
I know I can keep removing the sd card as required but it would be far simpler to be able to do it as designed through companion.

On another matter after updating to 2.3.11 only glider is shown on the new model screen on the transmitter.

Any help would to resolve would be appreciated.

3djc, 2021-02-27:
- if trims 1 and 4 pinched inward on power up do not start bootloader, then you need to reinstall it (please go to community support places like rcgroups or openrc forums for help on how to do that)
- the wizard issue is due to your Mac fiddling on the SDCard, you need to use CleanEject utility on your mac

closed 2021-02-27 by 3djc

Gadget99, 2021-02-27:
Hi,
Are you referring to reinstalling companion or bootloader?

3djc, 2021-02-27:
the opentx botloader on the radio

Gadget99, 2021-02-27:
Ok thanks. Ill give it a try