opentx/opentx #8330: Is it possible that OpenTX can have problems accessing External module

MartyHugh, 2021-02-27:
I am running two radiomaster TX16S each with a TBS Crossfire Micro.
They have identical models loaded, amix of CrossFire and Frsky (internal LBT 16) loaded.
One model has all the Frsky models cloned from the other.
So to all intents an purposes they are identical.

I am seeing an intermittent fault with both, the chances of this being hardware are limited in that both radios behave in a similar way.

The fault is when I have a Crossfire (External RF) model loaded sometimes I get the "Waiting for Crossfire" warning flashing indefinitely and the external module cannot be found.  The external module is powered up OK and binds straight away to the Crossfire Receiver, But the radio cannot communicate with it.

As I say, the fault is intermittent it works most of the time,  But when it does fail, the only thing which snaps it out of this failure mode is to switch the radio to a FrSky (i.e. internal RF) model, and then switch back to the Crossfire (External RF) module.

The fact that this reset works every time makes me wonder whether this is a software logic issue.

I would be grateful for any insights on what the cause might be, or whether there are any known or suspected problems switching between Internal and External

Thanks for reading.

3djc, 2021-02-27:
Could you try nigthlies ? Think we fixed it already

MartyHugh, 2021-02-27:
Thanks 3djc - will do.

Can you explain the cause of the problem so I can verify it fits my symptoms.

Thanks

MartyHugh, 2021-02-28:
Slightly awkward.... I have installed 2.3.12N448 companion.  How do I install firmware later than 2.3.11?  That is the latest the companion will download.  Is there a nightly option within companion somewhere ?

Thanks

3djc, 2021-02-28:
You need to set you channel to nightlies in setings/settings/application settings

MartyHugh, 2021-02-28:
Thanks. Doh!

MartyHugh, 2021-02-28:
Ok I have one radio with 2.3.12N448 and one with 2.3.11.

Can not induce the problem so far.  As the problem is intermittent though, it is a bit difficult to prove conclusively that it is no longer a problem in the nightly build.

From the fix(es) that were done in this area, can you give me some idea of how I can induce the fault on 2.3.11 so I can then do the same on the nightly to show the fault is no longer present.

Thanks

closed 2022-04-17 by stale[bot]