opentx/opentx #8331: Open TX 2.3.11 issues

Gadget99, 2021-02-28:
Since upgrading to 2.3.11 the model scripts have gone for plane and glider.
Also when trying to set up a new plane manually when not using the scripts there is no longer an internal RF listed only an external RF.

Gadget99, 2021-02-28:
Not sure if it helps but my TX is a Radiomaster TX16s. Firmware OpenTX-t16. Version 2.3.11-otx (4328055e). MacBook pro.

3djc, 2021-02-28:
- On Mac you need to use a utility like CleanEject to not corrupt the SDCard
- More than anything else, you need to flash the right firmware for your radio ....

closed 2021-02-28 by 3djc

Gadget99, 2021-02-28:
Been using cleanmydrive2 since release.
The firmware I am using is the latest. It was only flashed after having this and bootloader issues. Its shown as OpenTX-t16 once installed on the TX but the actual file is opentx-tx16s-en.

3djc, 2021-02-28:
No it does not

[image]

You have  the wrong firmware on your radio which does lead to your bootloader issue.