opentx/opentx #8337: lcd.drawBitmap(bitmap, x, y [, scale]) extend to lcd.drawBitmap(bitmap, x, y [[, scale],BmpOffsetX,BmpOffsetY])

Storchy1, 2021-03-04:
Perhaps it is possible to add these to parameters to the lua function.
I have seen that the luaLcdDrawBitmap in api_lcd.cpp 
**_static int luaLcdDrawBitmap(lua_State *L)_**
is almost ready for these two params.
If using the scale factor in lua script , the Offset of the bitmap is always set to (0,0).

Would be nice to have

Thank you

lshems, 2021-03-04:
Just for info: create function yourself in LUA:

LCD.drawBitmapRel=function(bitmap, x, y , scale,BmpOffsetX,BmpOffsetY)
  Return LCD.drawBitmapRel(bitmap, x-BmpOffsetX, y-BmpOffsetY, scale)
End

lshems, 2021-03-04:
Or even replace.

Storchy1, 2021-03-04:
I have tried this, i want to draw the bitmap screen(0,0) with offset (10,0)
so the function is LCD.drawBitmapRel(bitmap,-10,0, scale). 
This crash my widget.

lshems, 2021-03-04:
I use it all the time. Is it for widgets?

Storchy1, 2021-03-04:
Yes for widgets
In other words i want to draw a bitmap in the left upper corner of the x12s screen.
The bitmap is 500x 400 for instance. 
At the position screen(0,0) i want the bitmap position (100,0)

lshems, 2021-03-05:
check https://www.rcgroups.com/forums/showpost.php?p=38535221&postcount=1

from post 14 and on it is exactly explained how to do this.

and here a snippet:
`local zone = ...

--the widget specific lcd function container, passed as a global variable, to be passed to the script
local LCD = {}

zone=zone --if the size of the output zone is not specified
  or {x=0,y=0,w=LCD_W,h=LCD_H} --the entire screen of the current radio is taken.

--create any specific functions to redefine existing ones here
LCD.drawText = function(x,y,t,f)
  lcd.drawText(zone.x+x,zone.y+y,t,f)
end
  
--add any other generic lcd function you still need here
LCD.W=zone.w
LCD.H=zone.h

LCD.clear     = function() end

return {LCD=LCD}`

Storchy1, 2021-03-06:
I have change the title of the feature request. It was misunderstanding.
The request was a possibility to cropping the bitmap

Storchy1, 2021-03-10:
i have compiled my own 2.3.11 from the release tree for testing the function
i have changed the 
static int luaLcdDrawBitmap(lua_State *L)
{
  if (!luaLcdAllowed)
    return 0;
  const BitmapBuffer * b = checkBitmap(L, 1);

  if (b) {
    unsigned int x = luaL_checkunsigned(L, 2);
    unsigned int y = luaL_checkunsigned(L, 3);
    unsigned int scale = luaL_optunsigned(L, 4, 0);
    unsigned int bmpX = luaL_optunsigned(L, 5, 0);
    unsigned int bmpY = luaL_optunsigned(L, 6, 0);
    unsigned int bmpW = luaL_optunsigned(L, 7, 0);
    unsigned int bmpH = luaL_optunsigned(L, 8, 0);
    if (scale) {
      lcd->drawBitmap(x, y, b, bmpX, bmpY, bmpW, bmpH, scale/100.0f);
    }
    else {
      lcd->drawBitmap(x, y, b);
    }
  }
  return 0;
}
It works great. Perhaps it is possible to add the parameters to the lua function
Thank you very much

lshems, 2021-03-10:
How do I need to read this now? The offsets do exactly what? They define the opposite corner from the original bitmap? For cropping you need to define two opposing corners within the original, normally expressed in percentage to prevent invalid options.

Then scale determines if you want to adjust the size.

The X,Y tell you where you want that result to be drawn.

I'm missing something?

Storchy1, 2021-03-10:
Cropping the bitmap, bmpX and BmpY are the left upper corner in the bitmap and bmpW is the size in x ( width) and bmpH is the size in y ( Height).
Drawing a bitmap on screen on position (x,y) and the Bitmap offset is (bmpX/bmpY) and size to drawn is (bmpW x bmpH)
For instance the bitmap has the size (300x200), but i only want to draw 100x50 of the bitmap of the right/down corner of the bitmap, so 
lcd.drawBitmap(0,0,100 ,200,150,100,50)

Storchy1, 2021-03-10:
I have updated the title
Then i looks like this:

[image]

lshems, 2021-03-10:
Super nice. Opens interesting GPS applications

closed 2022-04-17 by stale[bot]