opentx/opentx #8340: No option in menu for failsafe

Gadget99, 2021-03-06:
I am setting up my first model on a TX16s. I’m using a cheap and cheerful model that uses internal RF Multi V761. Under the menu there is no option for failsafe shown. Is this due to the RF I’m using or have I missed a setting out to show it?
I’m using 2.3.11.

[image]
[image]

3djc, 2021-03-06:
Not all protocol allow failsafe settings on the radio. The information if supported or not comes from the multi directly, nothing we can do

closed 2021-03-06 by 3djc

Gadget99, 2021-03-06:
Ok thanks for the confirmation