opentx/opentx #8343: [BUG?] Incorrect decoding of Gas Suite temps, residual percent and flow data

pawelsky, 2021-03-07:
**Describe the bug**
According to FrSky S.Port protocol specification (version 3.4.7) following values are encoded on bits 0-15 only:
- Temperature1
- Temperature2
- Residual percent
- Flow
- Max flow
- Average flow

however OpenTx treats them as 32 bit values. This will cause displaying incorrect values if bits 16-31 are set to anything else than zeros for positive values or anything else than ones for negative values (e.g. for temperature). Is that a bug in OpenTx or FrSky spec? Has this been verified with an actual Gas Suite sensor sending negative temperature values?

**To Reproduce**
To reproduce the issue one needs to send temperature, residual percentage or flow data with e.g. 0x5555 in bits 16-31.

**Expected behavior**
Assuming FrSky spec is right data should be displayed correctly regardless of value of bits 16-31

**Screenshots**
Attached screenshot shows Temperature1 displayed as 65526 instead of -10 when 0x0000FFF6 is sent
[image]

**General information**
 - OpenTX 2.3.10
 - Taranis X9D

pawelsky, 2021-03-12:
Are there maybe any GasSuite owners here that could verify the displayed value for negative temperatures?

3djc, 2021-05-12:
Thanks for your thoughrough checks !