opentx/opentx #8345: When "Sensor Lost", SD Card log immediately changes schema, perhaps twice.

marked23, 2021-03-08:
FW    : opentx-x9d+2019
VERS : 2.3.7 (8b080683)
DATE : 2020-03-28 10:52:48
EEPR : 219

Internal Module
ISRM-S-X9
1.1.0/2.1.0FCC

Receiver
RX4R - ACCESS firmware v2.1.0

Receiving telemetry data from IKON2 (aka. BRAIN2) flight controller, over F.Port.  
CH-3 is a signaling port on the IKON2 that can talk F.Port (for telemetry) and S-Bus (for control) at the same time.
[image]

**Description:**
An interruption in sensor telemetry data causes the log file to change schema.
Some on-board sensors (local to the RX) reset, causing a second schema change in the log file. 

**Repro steps:**
Establish a setup where:
1. The transmitter is set to log telemetry to the SD card (0.5s)
the RX4R is receiving sensor telemetry from an external device over F.Port
the RX4R is sending S-Bus commands to the external device over the same F.Port  
(In my case, this is the IKON2)
1. Do something to interrupt or invalidate the F.Port signal until the TX cries "Sensor Lost"
1. Restore whatever interruption, so that F.Port might resume sensor telemetry
Let this state continue for a minute, to give you a chance to repro the second schema change 
1. Power everything down
1. Review the logs on the SD card

**Observed:**
* When sensor data disappears, the related columns also disappear from the logs
in my case: Tmp1, FLR, A2, RSSI, RxBt, Tmp2, RPM, RB1C, RB2C, VFAS, Curr
* Some of those columns come back
RSSI, RxBt (and maybe RB2C, but I'm not sure)

_header line added for context_
[image]

**Expected:**
* When sensor telemetry data disappears, the log schema remains unchanged.
_(Some non-value serves as a placeholder for no-data csv columns)_
* When sensor telemetry data re-appears, the log schema remains unchanged.
* When new sensors are discovered, the log schema remains unchanged until the TX is power-cycled.
* When the TX is power-cycled, the available sensors are compared to the existing log file schema... 
* If there is a mismatch, a new log file is started with the new schema.

kilrah, 2021-03-08:
First step would be to load current version and check the problem is still present, 2.3.7 is a year old.

mpjf01, 2021-03-29:
Seems to be possibly related to #7752

closed 2022-04-16 by stale[bot]