opentx/opentx #8347: Companion will crash and exit while editing Models Inputs when pressing Ctrl+A if no line is selected

wrdouglass, 2021-03-08:
**To Reproduce**
1. Select properties for any Model
2. Click on 'Inputs' tab
3. Select any row
4. Press Ctrl+A
5. Edit dialog will be displayed
6. Enter data as desired
7. Click OK
8. Press Ctrl+A (again)
9. Companion will hang briefly then exit, no error details are displayed, all changes are lost

**Expected behavior**
Step 8 should not cause Companion to crash and exit. It should ignore the Ctrl+A as there is nothing it can do in that scenario.

Note: Pressing Cancel in step 7 will also cause the crash in step 8

**General information**
 - Found in OpenTx Companion 2.3.11

wrdouglass, 2021-03-08:
I have code fixes for this and a couple similar issues in InputsPanel & MixesPanel but appears i don't have permission to push a branch for a pull-request.  Let me know how i can proceed here, i've looked around but i can't seem to locate any documentation on the steps for this project. Thanks.

3djc, 2021-03-08:
You can make PR out of fork branches

kilrah, 2021-03-08:
https://github.com/opentx/opentx/wiki/OpenTX-Git-Development-Workflow

wrdouglass, 2021-03-08:
@kilrah thank you.

elecpower, 2021-03-10:
Closed by merging PR

closed 2021-03-10 by elecpower